lgsvl/simulator #688: How to enable Radar in Apollo-5.0

ningfeiwang, 2020-03-18:
Hello,

I wonder how to enable radar for perception in Apollo-5.0 with LGSVL. In the json file, I file in the radar configurations. But when I turn on the button of Radar in Apollo-5.0 dreamview, It will turn off automatically. Could you please help me to deal with that?

EricBoiseLGSVL, 2020-03-18:
@luqiang21 does dreamview need any extra steps for radar?

luqiang21, 2020-03-19:
@EricBoiseLGSVL We didn't use Radar for Apollo's running.
@ningfeiwang I believe our simulator didn't simulate the raw radar data, what we simulated is the data after radar driver. So you don't need to enable the radar button on apollo. Currently, it is similar to lidar and camera, you don't need to enable their drivers in apollo as well to get sensor data in apollo.

ningfeiwang, 2020-03-19:
> @EricBoiseLGSVL We didn't use Radar for Apollo's running.
> @ningfeiwang I believe our simulator didn't simulate the raw radar data, what we simulated is the data after radar driver. So you don't need [...]

Thanks for the reply. I just followed what you said (not enable the driver of Radar) and added some logging in the perception modules. Then, I use the perception.sh to launch LiDAR, camera and Radar perception. But, it seems like that the Radar perception pipeline does not have any outputs. I wonder what the Radar data you simulated is used for. Will it use for Obstacle Detection? Thank you very much.

luqiang21, 2020-03-19:
@ningfeiwang I remember we simulated Radar for Apollo's navigation mode to use. We never tried to use it for standard mode. You can try to visualize radar output using rviz I believe. I don't have too much exposure to how Apollo's perception works with radar data yet.

closed 2020-04-08 by EricBoiseLGSVL

ningfeiwang, 2020-06-29:
Hello @luqiang21, I wonder what is the difference of RADAR simulation in LGSVL between navigation and standard mode. I check the Apollo source code. It seems that in the perception, there is no navigation flag. Seems that the navigation and standard modes are the same in perception.

luqiang21, 2020-06-29:
@ningfeiwang Our simulator doesn't have any difference in terms of implementation of RADAR for those modes in apollo. Probably you need to ask Apollo community since we don't have experience running Apollo with RADAR in standard mode. You might need to place radars according to their configurations.