lgsvl/simulator #690: Looping through waypoints

vatna, 2020-03-19:
I have given a NPC a list of waypoints, and all waypoints are set to the same speed. I am using `npc.follow(waypoints, loop=True)` to loop through them continuously. When it reaches the last waypoint, the vehicle goes back to the first waypoint at the same speed.

But what I'd like is that when it reaches the last waypoint, it teleports instantly back to the first waypoint and then goes through all the waypoints as normal until it reaches the last one, then teleports back, etc. Could you let me know how to do this?

vatna, 2020-03-20:
I've also tried to teleport the NPC by setting the state property of the NPC but it goes to the position at the same speed as it was moving instead of being instantly teleported there.

EricBoiseLGSVL, 2020-03-23:
@vatna This might require a few changes in Simulator and API.  I'll get back to you on this soon.

closed 2020-03-24 by vatna