lgsvl/simulator #691: The simulator and API communicate failed

liiiu, 2020-03-20:
hi, lg team

I tried running LGSVL in a LAN environment and my host IP was 10.20.180.212.  So, I changed the hostname and api_hostname in the config.yml file.

When I started LGSVL server, I was able to access webUI through 10.20.180.212:8080,

But I failed when I used the python API to communicate, I found that my listener port state looked like this.  

[fe80::a0b2:6075:3ff1:b198%5]:8181

I think it should be 10.20.180.212:8080. 

Do you know why? Thanks

ntutangyun, 2020-03-20:
@liiiu 8181 is the default API port, unless it's changed in simulator's config file. 

Could you check if you need to whitelist the port `8181` in firewall with `ufw`.

liiiu, 2020-03-21:
I don't think it's the firewall.

I found the problem in `ApiManager.cs`

```
        void Awake()
        {
            roadLayer = LayerMask.NameToLayer("Default");

            DontDestroyOnLoad(gameObject);
            Instance = this;

            IPAddress address;
            if (Config.ApiHost == "*")
            {
                address = IPAddress.Any;
            }
            else if (Config.ApiHost == "localhost")
            {
                address = IPAddress.Loopback;
            }
            else
            {
                var entries = Dns.GetHostEntry(Config.ApiHost);
                if (entries.AddressList.Length == 0)
                {
                    throw new Exception($"Cannot resolve {Config.ApiHost} hostname");
                }
                address = entries.AddressList[0];
            }

            Server = new WebSocketServer(address, Config.ApiPort);
            Server.AddWebSocketService<SimulatorClient>("/");
            Server.Start();
            SIM.LogAPI(SIM.API.SimulationCreate);
        }

```

We can see that the code does not do IP filtering after DNS resolution

[image]

EricBoiseLGSVL, 2020-10-06:
This is an issue with web socket library not supporting ipv6 so we made a fix for the next release

closed 2020-10-06 by EricBoiseLGSVL