lgsvl/simulator #693: Shader error

JockTokk, 2020-03-21:
Hi, LG Team, I have a problem with shader in SanFrancisco map. I've used `git lfs install`. 

[image]

Shader error in 'Shader Graphs/EnvironmentSimulation': 'NewGradient': no matching 0 parameter function at line 1537 (on d3d11)

Compiling Vertex program with UNITY_PASS_META
Platform defines: UNITY_ENABLE_REFLECTION_BUFFERS UNITY_USE_DITHER_MASK_FOR_ALPHABLENDED_SHADOWS UNITY_PBS_USE_BRDF1 UNITY_SPECCUBE_BOX_PROJECTION UNITY_SPECCUBE_BLENDING UNITY_ENABLE_DETAIL_NORMALMAP SHADER_API_DESKTOP UNITY_LIGHT_PROBE_PROXY_VOLUME UNITY_LIGHTMAP_FULL_HDR

Shader error in 'Shader Graphs/EnvironmentSimulation': syntax error: unexpected integer constant at line 1537 (on d3d11)

Compiling Vertex program with UNITY_PASS_META
Platform defines: UNITY_ENABLE_REFLECTION_BUFFERS UNITY_USE_DITHER_MASK_FOR_ALPHABLENDED_SHADOWS UNITY_PBS_USE_BRDF1 UNITY_SPECCUBE_BOX_PROJECTION UNITY_SPECCUBE_BLENDING UNITY_ENABLE_DETAIL_NORMALMAP SHADER_API_DESKTOP UNITY_LIGHT_PROBE_PROXY_VOLUME UNITY_LIGHTMAP_FULL_HDR

Shader error in 'Shader Graphs/EnvironmentSimulation': 'NewGradient': no matching 0 parameter function at line 4839 (on d3d11)

Compiling Vertex program
Platform defines: UNITY_ENABLE_REFLECTION_BUFFERS UNITY_USE_DITHER_MASK_FOR_ALPHABLENDED_SHADOWS UNITY_PBS_USE_BRDF1 UNITY_SPECCUBE_BOX_PROJECTION UNITY_SPECCUBE_BLENDING UNITY_ENABLE_DETAIL_NORMALMAP SHADER_API_DESKTOP UNITY_LIGHT_PROBE_PROXY_VOLUME UNITY_LIGHTMAP_FULL_HDR

Shader error in 'Shader Graphs/EnvironmentSimulation': syntax error: unexpected integer constant at line 4839 (on d3d11)

Compiling Vertex program
Platform defines: UNITY_ENABLE_REFLECTION_BUFFERS UNITY_USE_DITHER_MASK_FOR_ALPHABLENDED_SHADOWS UNITY_PBS_USE_BRDF1 UNITY_SPECCUBE_BOX_PROJECTION UNITY_SPECCUBE_BLENDING UNITY_ENABLE_DETAIL_NORMALMAP SHADER_API_DESKTOP UNITY_LIGHT_PROBE_PROXY_VOLUME UNITY_LIGHTMAP_FULL_HDR

Shader error in 'Shader Graphs/EnvironmentSimulation': 'NewGradient': no matching 0 parameter function at line 7359 (on d3d11)

Compiling Fragment program with DECALS_3RT LIGHT_LAYERS
Platform defines: UNITY_ENABLE_REFLECTION_BUFFERS UNITY_USE_DITHER_MASK_FOR_ALPHABLENDED_SHADOWS UNITY_PBS_USE_BRDF1 UNITY_SPECCUBE_BOX_PROJECTION UNITY_SPECCUBE_BLENDING UNITY_ENABLE_DETAIL_NORMALMAP SHADER_API_DESKTOP UNITY_LIGHT_PROBE_PROXY_VOLUME UNITY_LIGHTMAP_FULL_HDR

Shader error in 'Shader Graphs/EnvironmentSimulation': syntax error: unexpected integer constant at line 7359 (on d3d11)

Compiling Fragment program with DECALS_3RT LIGHT_LAYERS
Platform defines: UNITY_ENABLE_REFLECTION_BUFFERS UNITY_USE_DITHER_MASK_FOR_ALPHABLENDED_SHADOWS UNITY_PBS_USE_BRDF1 UNITY_SPECCUBE_BOX_PROJECTION UNITY_SPECCUBE_BLENDING UNITY_ENABLE_DETAIL_NORMALMAP SHADER_API_DESKTOP UNITY_LIGHT_PROBE_PROXY_VOLUME UNITY_LIGHTMAP_FULL_HDR

EricBoiseLGSVL, 2020-03-23:
@JockTokk Unity shadergraph is still has some bugs.  Navigate to the shader and double click it.  This will open shadergraph editor.  In the top left, click Save Asset...  This will recompile the shader.  Let me know if this fixes the issue.

JockTokk, 2020-03-23:
Thank you! I will try it and I will report the result.

JockTokk, 2020-03-23:
@EricBoiseLGSVL, thank you! I will try it and I will report the result.

JockTokk, 2020-03-24:
@EricBoiseLGSVL, unfortunatelly, there are new errors.

[image]

EricBoiseLGSVL, 2020-03-24:
@JockTokk Ok, in project root, while Unity is closed, delete Library folder.  Then open Unity and it will rebuild this folder.  Let me know if this fixes the issues

JockTokk, 2020-03-25:
@EricBoiseLGSVL, I've done this instruction, but it hasn't helped.

[image]

EricBoiseLGSVL, 2020-03-25:
Can you please double click on shader graph EnvironmentSimulation, click on Save Asset... in the top left.  This will recompile the shader.  The other errors are fine.  You may need to run the HDRP wizard also but try recompiling

JockTokk, 2020-03-26:
I'am on Windows.
Unfortunatelly, all is the same

[image]

EricBoiseLGSVL, 2020-03-26:
@JockTokk This seems like code in the shader is not supported by your GPU or driver issues.  What GPU are you using and what driver version are you on?

martins-mozeiko, 2020-03-26:
This would be really strange if this would be GPU bug. My guess is on Unity messing up generated shader graph code.

@JockTokk Can you do following, please?

1) Double click EnvironmentSimulation shader (or click `Open Shader Editor` button from your last screenshot)
2) On the rightmost node called `Lit Master` click right mouse button and select `Show Generated Code`
3) Now scroll to line 1937 or 4839 or 7359 and copy & paste it here (maybe one extra line above and below too)

It could be this bug: https://github.com/Unity-Technologies/ScriptableRenderPipeline/pull/5424
If locale settings are messing up float conversion and instead 1.0 you get 1,0 - that's a syntax error. @JockTokk what system locale you have? Do you use , or . for decimal point? Can you try changing that in your system locale settings? When you change it, restart Unity to see if this gets fixed.

JockTokk, 2020-03-26:
@EricBoiseLGSVL, NVIDIA GeForce GT 720M. Driver version: 391.35.

EricBoiseLGSVL, 2020-03-26:
@martins-mozeiko thanks, yes this has been an issue before.
@JockTokk Is your GPU 2gb?  If so, you will have many issues running simulator.  We recommend at least 8gb.

JockTokk, 2020-03-26:
@martins-mozeiko, you are right. It's related with system locale.
Line number 9759:
`Gradient _Gradient_AFE042FB_Out_0 = NewGradient(0, 5, 2, float4(0, 0, 0, 0,08529793),float4(0,8490566, 0,8490566, 0,8490566, 0,6470588),float4(1, 1, 1, 0,7617609),float4(0,1037736, 0,1027946, 0,1027946, 0,9264668),float4(0, 0, 0, 1),float4(0, 0, 0, 0),float4(0, 0, 0, 0),float4(0, 0, 0, 0), float2(1, 0),float2(1, 1),float2(0, 0),float2(0, 0),float2(0, 0),float2(0, 0),float2(0, 0),float2(0, 0));`

I changed the comma to the dot, but it hasn't helped. I've tried to remove folder Library in project root. It also hasn't helped.

@EricBoiseLGSVL, yes. It's just test.

martins-mozeiko, 2020-03-26:
You changed it here? To dot?
[image]

And then closed Unity fully, including Unity Hub, and then reopened project again?

JockTokk, 2020-03-26:
> You changed it here? To dot?'

@martins-mozeiko, yes. 

> And then closed Unity fully, including Unity Hub, and then reopened project again?

Yes.

martins-mozeiko, 2020-03-26:
Maybe try logging out from your user and logging back on Windows. I don't remember how regional settings were applied in Windows.

JockTokk, 2020-03-26:
@martins-mozeiko I've already tried that. I've also tried to reboot system.

martins-mozeiko, 2020-03-26:
Alternative would be to open `GradientShaderProperty.cs` file and apply these changes manually: https://github.com/Unity-Technologies/ScriptableRenderPipeline/pull/5424/files#diff-c233a20539fb47175c19f9a89d09837c

This file is inside `Library/PackageCache/com.unity.shadergraph@5.16.1/Editor/Data/Graphs` folder. 

After that try reimporting the shader again.

JockTokk, 2020-03-26:
@martins-mozeiko I've just reinstalled project. Errors have disappeared. But...

[image]

In play mode

[image]

EricBoiseLGSVL, 2020-03-26:
@JockTokk Yes, in editor there is no light source.  In play mode we add the directional light.  To have lighting in scene view press the button top left of scene view panel.

[image]

JockTokk, 2020-03-27:
@EricBoiseLGSVL yes exactly.

JockTokk, 2020-03-27:
@EricBoiseLGSVL @martins-mozeiko thank you very much for the help!

For anyone, who will face this problem.

1. You need to change the decimal point in the system from the comma to the point.

[image]

2. Reboot your computer.

3. In project right click on EnvironmentSimulation shader and choose Reimport. It's in `Assets/Shaders/EnvironmentShaders`.

4. If it won't work, you need to again create new project.

5. If it won't work, you need to 

> Alternative would be to open `GradientShaderProperty.cs` file and apply these changes manually: https://github.com/Unity-Technologies/ScriptableRenderPipeline/pull/5424/files#diff-c233a20539fb47175c19f9a89d09837c
> 
> This file is inside `Library/PackageCache/com.unity.shadergraph@5.16.1/Editor/Data/Graphs` folder.
> Note that changes will disappear, if Library folder gets removed.
> 
> After that try reimporting the shader again.

6. At the end if you want to see city in scene, you need to toggled off this button

[image]

closed 2020-03-27 by JockTokk