lgsvl/simulator #694: Saturation the Steering Angle

ozguradas, 2020-03-21:
Hello all,

We have a vehicle on Unity, as you know there is a dynamic spec which is max. steering angle. When i changed it and rebuild, i cant see any differences on LGSVL. And then i found that line in vehicleDynamic.cs :

Tooltip("steering range is +-maxSteeringAngle")]
    public float maxSteeringAngle = 29.4f;

and as you see i changed it as 29.4f. But again it isn't working on Lgsvl. What am i doing wrong about that ? Any tips ? Thanks for feedbacks !

EricBoiseLGSVL, 2020-03-23:
@ozguradas when a variable in Unity C# classes are public, they are serialized and exposed in the editor inspector panel.  This inspector panel overrides the class initialized value.  This is a common issue when working with Unity.  Just select the prefab and change the value in the editor, then save the prefab

ozguradas, 2020-03-24:
Oh i got it, thank you Eric !

closed 2020-03-24 by ozguradas