lgsvl/simulator #695: can't load vector map

chronofei, 2020-03-23:
Hey guys, I am working with lgsvl 2020 and autoware 12. Now I can load pointcloud map, but not vector map, the topic /vector_map echo nothing, /vmap_stat echo false.

EricBoiseLGSVL, 2020-03-23:
@hadiTab  Do you know what the issue is here?

hadiTab, 2020-03-23:
@chronofei are you using the maps from our [autoware-data](https://github.com/lgsvl/autoware-data) repo?
If so, which map are you having problems with?
What steps are you following in Autoware? Do you get any errors when you load the maps?

chronofei, 2020-03-24:
@hadiTab thank you for your reply. I am using the maps from autoware-data repo, and I have installed git-lfs. Now I have pointcloud map, but the map doesn't match the lgsvl scenes, so that the pose that ndt matching give is wrong. And the vector map can't load. I try all the maps that autoware-data repo give. I load the maps via quick start, there is no errors echo. I can load officially map data.

hadiTab, 2020-03-27:
The pointcloud map should match the scene in the simulator. Which scene are you using in the simulator?

chronofei, 2020-03-29:
San Francisco, but I try all the scene and pointcloud map. Maybe I make some mistakes. I don't know how to deal with it.

Jiyunfeng1989, 2020-04-05:
i also have same problem. is there anybody who solve this. thanks a lot

chronofei, 2020-04-06:
@hadiTab  hey friend, can you solve this problem?

hadiTab, 2020-04-07:
@chronofei can you show me a screenshot of what things look like in rviz? I'm having a hard time figuring out the issue with information provided here.

Normally when the Autoware runtime manager is running it will print a lot of log messages in the terminal when you launch nodes (such as map loader). If there are any errors loading the map they will be printed in the terminal. If the map is loaded successfully it will also be noted there. Please keep that terminal in the foreground when you attempt to load the map to see what it prints out.

EricBoiseLGSVL, 2020-04-29:
@Jiyunfeng1989 @chronofei are you still having issues?

closed 2020-07-31 by lemketron