lgsvl/simulator #696: LGSVL's nmea_sentence doesn't change

eungh1, 2020-03-24:
Hello guys, I'm a student working with LGSVL wit Autoware.
I want to use LGSVL's nmea_sentence topic for localization.
So I change Map origin's parameter using Unity. But, it doesn't change.

I attach pictures.

[image]
[image]
[image]
These are information from the CubeTown map provided by LGSVL.

[image]
[image]  
[image]
And these are information from the CubeTown map, which has modified the Map origin parameter.

As you can see, Map origin was modified through Unity, but the information of nmea_sentence has not changed.
How can I change nmea_sentence?

Thanks.

EricBoiseLGSVL, 2020-03-26:
@eungh1 We have recently updated this sensor but not released the changes.  Let me see if @rongguodong can help

rongguodong, 2020-03-26:
@eungh1 Are you using Autowware.AI or Autoware.Auto? How do you display the nmea_sentence topic?
Can you open the GPS visualizer (by click the "eye" icon) to see if Longitude/Latitude there change when you change the map origin?

rongguodong, 2020-03-26:
@eungh1 I think I know why. If you are using our default vehicle setup for Autoware, the GPS sensor has: "IgnoreMapOrigin": true. Just change that to false to have map origin NOT ignored. :)

eungh1, 2020-03-27:
@EricBoiseLGSVL Thank you for kindness. :)

@rongguodong I'm using Autoware.AI and I display the nmea_sentence topic using just rostopic echo.
I changed the "IgonoreMapOrigin" to false as you told me and it solved the problem. Thank you so much!

eungh1, 2020-03-27:
It is different from the issue, but can I get information on the condition of the vehicle(not cmd, for example, steering angle)? It is okay to think that the vehicle moves immediately according to the cmd value?

EricBoiseLGSVL, 2020-03-27:
@eungh1 yes.  You can get data from CanBus sensor or create your own sensor that sends what data you need.  Look at canbus as an example on how to create.  Note, if you plan on making a new sensor that is built in simulator, be sure to add prefab to RuntimeSettings.asset sensor list

eungh1, 2020-03-30:
@EricBoiseLGSVL I know your default vehicle for Autoware.ai does not support CanBus sensor. If you support CanBus sensors for Autoware.ai, how can I use that?
Or do you mean to create a sensor that publishes the desired data? I don't know what example about canbus you are talking about. Is it found in the LGSVL document?

EricBoiseLGSVL, 2020-03-30:
@eungh1 any vehicle can support any sensor.  You just need to alter the JSON sensor configuration in the Simulator WebUI.  You can also build Simulator with a new sensor or build a sensor plugin to send the data you need.  This is in our documentation on our website and in the repo.

closed 2020-04-29 by EricBoiseLGSVL