lgsvl/simulator #698: Current Simulator and Lane following tutorial don't work

dr563105, 2020-03-24:
Hello!

I'm working on making the lane following tutorial work with the current Simulator release as the steering commands logic has changed since the version in which LF was developed to work(2019.5).

Lane following -- https://www.lgsvlsimulator.com/docs/lane-following/

I contacted David Uhm through email and he told me to look into how controlling ego vehicle part has changed (​​what's in a control message, which ROS topic it subscribes to, etc) and update the steering commands that we are sending from LaneFollowing ROS node accordingly. 

So I looked in and saw that [drive.py](https://github.com/lgsvl/lanefollowing/blob/master/ros2_ws/src/lane_following/drive.py) subscribes to `/simulator/sensor/camera/center/compressed` and publishes `/lanefollowing/steering_cmd`. The [collect.py](https://github.com/lgsvl/lanefollowing/blob/master/ros2_ws/src/lane_following/collect.py) subscribes to `'/simulator/sensor/camera/center/compressed'  '/simulator/sensor/camera/left/compressed'  '/simulator/sensor/camera/right /compressed'  '/simulator/control/command' ` 

In one of your [issues](https://github.com/lgsvl/simulator/issues/275), somebody had already asked about controlling and collecting data. You suggested to look into [VehicleController.cs](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Controllers/VehicleController.cs) as how the apollo and autoware process messages. I compared the [VehicleController_obsolete.cs](https://github.com/lgsvl/simulator-2019.05-obsolete/blob/master/Assets/Scripts/VehicleController.cs) file in an obsolete version of the Sim. On first glance, I see that the lines of code have been drastically reduced. That makes it little difficult to compare them. Can you please tell me which lines target this functionality?

Finally my questions are --
1. Am I in the right track to solving the problem and making the tutorial run with the current simulator? 
2. Since the instructions to run the sim is also outdated, could you please tell me what options to choose when I run the Simulator, goto `localhost:8080` and select `simulations` tab? 
3. When I run the command -- `docker-compose up drive` , although, the code is outdated, I see that `/usr/bin/rosbridge` is running in the background and consumes quite a lot of cpu(about 95%). I haven't made any changes to the `drive.py`. Please see here -- [https://imgur.com/PqRZmiE](https://imgur.com/PqRZmiE). Could you please tell me how manage it to use less resource? 

Thank you.

EricBoiseLGSVL, 2020-03-26:
@dr563105 VehicleController.cs collects inputs from all control types.  AD, keyboard, wheel, etc.  This is then read by VehicleSMI.cs to calculate and apply force to the wheel colliders.  Look at VehicleControlSensor.cs and follow AccelInput and SteerInput vars.

1) Yes, you are on the right track to run the tutorial.  We will need to update this in a future release.
2) Setup will just be including sensor parameters and bridge settings in vehicle settings instead of on prefab.  Set the Map to San Francisco.
3) I think an updated ros has fixed this cpu issue.  Right @hadiTab ?

I apologize that some of our older simulator documentation has not been updated.  We will update these as soon as we can.

dr563105, 2020-05-04:
Hey @EricBoiseLGSVL, is there a topic similar to /simulator/control/command which publishes control commands to the outside, present in the current sim version? I'd like not use Autoware or Apollo to fetch this info. 
If not, how do I go about adding a topic, sensor to the sim files?

daviduhm, 2020-05-04:
Hi @dr563105, we've been discussing this via email some time ago, and I just wanted to let you know that recently I've been working on updating the LaneFollowing repo and adding a sensor for it to work with the latest simulator. I'll be pushing the updates soon today or tomorrow and back to you with some instructions. Sorry that it took longer than I expected.

daviduhm, 2020-05-05:
@dr563105 I've updated the docker image for Lane Following to use ros2 dashing and fixed the ros2 web bridge build issue. Please pull the latest (https://github.com/lgsvl/lanefollowing) and build again (notice that build command for ros2 packages is now `docker-compose up build_ros`).

I've also implemented a custom sensor plugin for this Lane Following model (https://github.com/lgsvl/LaneFollowingSensor). Please see the sensor plugins documentation on our website (https://www.lgsvlsimulator.com/docs/sensor-plugins) and the readme instructions on this repo for how to set it up and use. You'll also need to patch the latest simulator with these two patch files as I've needed to fix some bugs for this to work. These patches will also be pushed to the master later on, but I'm just including them here so you can test it now.
- https://drive.google.com/open?id=19ZC_Ty4uqCqiwXMzVJdgti8N5dRk_ani
- https://drive.google.com/open?id=1zvxtQVkYpM2Udocsg-bdyFzsYdvhCmcQ

For testing, I've used our Single Lane Road map (https://content.lgsvlsimulator.com/maps/singlelaneroad/) as this map doesn't have any intersections.

[image]

dr563105, 2020-05-05:
Thanks a lot David. This is excellent news. I will get to testing right away.

dr563105, 2020-05-06:
@daviduhm 
Update 2: I solved getoutputtype error. I cloned release 2020.03 branch instead of master. IDataConverter.cs had a definition for the type. I added the line and build was successful. 
I was able to run the data collection part of the tutorial but not autonomous part. When I do `docker-compose up drive_visual` and select `main camera and lanefollowing sensor`, the topic `/lanefollowing/steering_cmd` stays at **count:0** in the bridge menu in the Simulator. Hence the vehicle doesn't move autonomously. What do you suggest?

daviduhm, 2020-05-07:
Which version/branch of simulator are you using? Supporting custom sensor plugins and custom sensor messages are quite new features. Please use the latest master branch or custom sensors might not work. I've tested those two patches working fine on the latest master.

As for the dockerfile, goal was to fix ros2 bridge issue and the solution was to update ros2 to dashing. Fixing version numbers for python packages were to make sure that the docker image uses the same version of those packages every time we build so it won't break anything. Upgrading CUDA version or tensorflow wasn't necessary and the current versions are fine for running the lane following model.

A custom sensor plugin and custom sensor are the same concept. If you were to create another custom sensor that publishes your custom messages, you'll have to write a custom sensor with custom sensor messages following this documentation: [https://www.lgsvlsimulator.com/docs/sensor-plugins/](https://www.lgsvlsimulator.com/docs/sensor-plugins/) Please also take a look at how I've implemented Lane Following sensor as an example: https://github.com/lgsvl/LaneFollowingSensor

EricBoiseLGSVL, 2020-05-07:
@dr563105 We are looking at why that bundle fails to download.  Should be fixed soon

daviduhm, 2020-05-07:
As for the download for Lincoln MKZ, it was missing the vehicle name at the end of the url. It's fixed now: https://content.lgsvlsimulator.com/vehicles/lincolnmkz2017/

Thanks for pointing it out.

daviduhm, 2020-05-07:
As for the download link for Lincoln MKZ, it was missing the vehicle name at the end of the url. It's fixed now: https://content.lgsvlsimulator.com/vehicles/lincolnmkz2017/

Thanks for pointing it out.

dr563105, 2020-05-11:
> Which version/branch of simulator are you using? Supporting custom sensor plugins and custom sensor messages are quite new features. Please use the latest master branch or custom sensors might not work. I've tested those two patches working fine on the latest master.

Well it was suggested to use 2020.03 release instead of master. Hence I did it that way. Only by looking around and comparing, I found the missing line. https://github.com/lgsvl/simulator/releases/tag/2020.03

> As for the dockerfile, goal was to fix ros2 bridge issue and the solution was to update ros2 to dashing. Fixing version numbers for python packages were to make sure that the docker image uses the [...]

OK. I tried to change CUDA and TF to the latest and faced errors multiple times while building a docker image by myself which took ages to complete. Hence wondered if it is possible to have an image with all the latest version modules. No worries. 

> A custom sensor plugin and custom sensor are the same concept. If you were to create another custom sensor that publishes your custom messages, you'll have to write a custom sensor with custom [...]

Thank you David. I learned so much just by seeing your implementation, comparing to ComfortSensor and also looking at RosConversions for further extension. 
I created a custom message, added it to lgsvl_msgs package, added it to src/ of the Lanefollowing project. Then proceeded to add the class in LGSVL.cs. After, I extended the LaneFollowingData and Sensor files to my needs. However when I try to build the simulator in Unity I'm getting compile errors such as `Assets/External/Sensors/LaneFollowingSensor/LaneFollowingSensor.cs(55,44): error CS1061: 'LaneFollowingData' does not contain a definition for 'twist' and no accessible extension method 'twist' accepting a first argument of type 'LaneFollowingData' could be found (are you missing a using directive or an assembly reference?)`

If it is not much trouble, could you please have a look at my files and point me out how I make it right? Also if possible, please, tell me if my approach to extension of the plugin is right and possibly how it could be made better or cleaner? 

https://gist.github.com/dr563105/3bd38e12febf0b87a2eda8f345b2fde9

dr563105, 2020-05-14:
Thanks @daviduhm @EricBoiseLGSVL . I used the release branch as it was suggested to use it instead of master. No worries using master branch now. 

After training the lanefollowing model I test it by executing `docker-compose up drive_visual`. In the previous old simulator, I had to select the main camera to start the autonomous drive mode display window. However in this, all I have to do is press `play` in the simulator, the auto drive window pops up immediately. I could see the steering input prediction in the terminal. But the car doesn't move forward on its own in both auto drive window and simulator. Is this the expect behaviour? 

In the sensor plugin -- `LanefollowingData.cs` if I have to return a message type something else including TwistStamped messagetype, how could I implement it? 
https://github.com/lgsvl/LaneFollowingSensor/blob/master/LaneFollowingData.cs#L22-L56
And read more than TwistStamped as here -- https://github.com/lgsvl/LaneFollowingSensor/blob/master/LaneFollowingSensor.cs#L46 ?

I referred to [ComfortSensor](https://github.com/lgsvl/ComfortSensor). There are two message types defined and one is [returned](https://github.com/lgsvl/ComfortSensor/blob/master/ComfortData.cs#L49-L70). It doesn't include one or more built-in message types in Ros.cs. Hence I'm stuck on creating my own sensor plugin. In fact I don't understand the logic behind such approach. Could you please explain why you do this? 

Thank you.

dr563105, 2020-06-11:
@EricBoiseLGSVL @daviduhm do the changes mentioned here -- https://github.com/lgsvl/simulator/issues/698#issuecomment-623810249 integrated to the latest sim? I'd like to add the sensor and do some work on it.

daviduhm, 2020-06-12:
@dr563105 Yes, those issues were fixed on the latest master as well as 2020.05 release branch.

dr563105, 2020-07-13:
Hey @daviduhm, some weeks I tried to use the 2020.05 version and since the whole display got darker when hit `play`, I left it as it was and continued with 2020.03. Last week I came back to 05 and tried to verify the changes. I see that the changes to 
https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Bridge/Cyber/CyberWriter.cs,
https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Bridge/IWriter.cs,
https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Bridge/Ros/RosWriter.cs,
https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Bridge/Ros/RosPointCloudWriter.cs 
have not been updated. However, the lanefollowing sensor worked when I tried it. So are these changes necessary? 

Upgrading to 2020.05 and changing these files, got me dependency errors. I don't want to go into the rabbit hole to fix each changes. What do you advise?

closed 2020-10-20 by EricBoiseLGSVL

AlexandrZabolotny, 2022-10-04:
Hello everyone!
I'm trying to run a simulator for lane following. I want to publish a topic for driving ego car
I have version 2020.06
I am following the advice above. Add this [repo](https://github.com/lgsvl/LaneFollowingSensor)
When I try to build the simulator I get the following errors.
Maybe someone has some ideas?

[image]

EricBoiseLGSVL, 2022-10-05:
Be sure you are on release 2020.06 branch in simulator.  Master is behind releases and the repo you cloned is on the 2020.06 tag

EricBoiseLGSVL, 2022-10-07:
is the sensor repo on 2020.06 tag?

AlexandrZabolotny, 2022-10-10:
> is the sensor repo on 2020.06 tag?

oh no. I thought the sensor has only one branch.
I've switched to the required version and now it works.
Sorry for disturbing