lgsvl/simulator #699: Creating a new map

Erdemhan, 2020-03-26:
Hi , i apologize if the question is already answered or if it is in wrong place.

I am new at lgsvl and unity we are trying to build a simulation for our autonomus vehicle using pythonAPI. 
Question is we need to build our own map , actually it is a basic racecourse. We have 3d models of this racecourse i find some information about map annotations but i dont know how to create a lgsvl map with this models.

I apologize for my English if there is a mistake.

EricBoiseLGSVL, 2020-03-26:
@Erdemhan 
Welcome to LGSVL Simulator.  We are working on updating our documentation on creating user content so you are right to have questions.

Since this is an advanced topic, I'll need to break the steps up.

1) Make sure the track model is game ready.  If the model is millions of polygons, you are going to have a terrible frame-rate.  Please look at game ready models requirements in Unity tutorials.

2) Make sure you are compressing textures.  Work your way from 1k to 4k texture compression depending on fps.

3) Look at how CubeTown folder structure is setup in Assets/External/Environments/CubeTown.  This is how you will need to setup your new map.  Import Meshes and textures to folders here.  Move any third party assets here as well or checks will fail.

4) We have edited Unity's HDRP code when creating a new scene so it adds the required game objects when creating a new scene with the editor.  Create a new scene, and save it as the same name as the folder structure.

5) Create the scene with your assets.  Be sure to look at CubeTown again to see the scene structure.  MapOrigin, Environment, MapHolder, etc.  I'd recommend making prefabs of objects to help with source control.  Also, be aware that modeling software can have different scales so be sure importer in Unity editor sets assets to the right size.  Place a vehicle or a simple capsule to make sure the scale is correct.  Start simple and decorate after everything is working.

6) Create mesh colliders for roads and simple colliders for buildings and obstacles.  Be sure the meshes are Read/Write enabled in the importer again.

7) Set tags for semantic camera is needed.  Remember that you need to set parent objects because the code traverses children and doesn't support multiple tags per mesh.

8) Rename MapHolder to Map"SceneName" and begin annotating.  Our docs will help on how to do this and so will looking at MapCubeTown.prefab to see how it is structured.

9) Click Simulator -> Build... to open the build options.  Here you create a bundle of your new map so it can be loaded.  After building, open WebUI and set the path to a new environment and you are ready to run.

This is a broad list, so let me know if you have issues or need more details.  Remember to balance visuals with performance.

Erdemhan, 2020-03-27:
Thanks a lot , it worked fine for me.
only problem that i have is:
`Building an AssetBundle for target 'LinuxStandaloneSupport' is not allowed because the required module is not installed`
and i found the solution here:
#475 

Now i have another question.
Is there any possibility for limit simulation fps?
Because in this way i have 200+ fps and %99 gpu usage in simulation and its really disturbing. Only 60 fps is really fine for me. 

Thanks again for answer.

closed 2020-03-27 by Erdemhan