lgsvl/simulator #700: How to modify the speed of NPC vehicles during simulator running

zhwmonkey, 2020-03-27:
Hi, I want to create a scene which the speed of NPC can change suddenly. Is there any way to do it?
Thanks

EricBoiseLGSVL, 2020-03-27:
@zhwmonkey How do you plan on sending the command to change speed?

zhwmonkey, 2020-04-02:
I'm so sorry for replying so late. I'm going to write a python script to create a scenario that contains the initial and triggered behavior of the ego vehicle and NPC.For example, when the distance between the ego vehicle and the NPC is less than 100m, the NPC slows down to the specified speed.
Thanks a lot.

EricBoiseLGSVL, 2020-04-02:
@zhwmonkey No worries.  I think this will be easier by creating custom behaviors for NPC's.  We are currently refactoring and making extensional the NPC behavior class, NPCController.cs.  This should release in April.

In the meantime, I suggest creating a custom sensor with a callback for all NPC's that are within range.  Then the python api cycles through this list and sets the NPC speed.  NPCManager.cs has all active npc's in a list to iterate through in the new sensor.

closed 2020-10-20 by EricBoiseLGSVL