lgsvl/simulator #705: Kernel 'KSampleCopy4_1_x_8' not found while trying to build standalone executable

tleyden, 2020-03-30:
I've tried to follow the [Instructions to build standalone executable](https://www.lgsvlsimulator.com/docs/build-instructions/), but running into errors in the Unity Editor:

[image]

I did make sure to use the version of Unity mentioned in the docs:

[image]

with Windows support:

[image]

I tried clearing the errors, but they re-appeared after running the git clone commands to pull in the vehicle and environment assets:

```
$ git clone https://github.com/lgsvl/CubeTown.git
$ git clone https://github.com/lgsvl/Jaguar2015XE.git
```

[image]

I also tried to run Simulator/Check, which gave me these errors:

[image]

## System details

* System76 Oryx Pro with Geforce 1070 (8 GB GPU RAM)
* Linux 4.15.0-88-generic #88-Ubuntu SMP Tue Feb 11 20:11:34 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
* nvidia-smi output

```
# nvidia-smi
Sun Mar 29 09:45:54 2020
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 418.56       Driver Version: 418.56       CUDA Version: 10.1     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce GTX 107...  Off  | 00000000:01:00.0 Off |                  N/A |
| N/A   43C    P2    30W /  N/A |    136MiB /  8119MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+

+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|    0     11110      C   /usr/NX/bin/nxnode.bin                       126MiB |
+-----------------------------------------------------------------------------+
```

tleyden, 2020-03-30:
After re-installing the nvidia drivers to:

```
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 440.59       Driver Version: 440.59       CUDA Version: 10.2     |
|-------------------------------+----------------------+----------------------+
```

based on the advice in https://github.com/lgsvl/simulator/issues/287#issuecomment-524404393, it's getting past the KSampleCopy4_1_x_8 errors, and in fact I haven't seen any errors so far while opening the project in the Unity Editor.

[image]

closed 2020-03-30 by tleyden

ChaoqinRobotics, 2021-12-08:
I follow this video https://www.youtube.com/watch?v=VqDFFyQRj8E and solve the problem.