lgsvl/simulator #708: Running the LGSVL simulator alongside Autoware.Auto using VNC

PunchShadow, 2020-03-31:
Hi everyone,

I try to follow this [tutorial](https://autowarefoundation.gitlab.io/autoware.auto/AutowareAuto/lgsvl.html) to open LGSVL simulator.
I execute this on server with Ubuntu 16.04 LTS, GTX1080ti and already installed nvidia driver passing GUI by VNC.
Everything goes well till I choose the resolution and click OK

[image]

However, It turns out to miss the display and the error is:
[image]

So I've wonder whether I can use VNC to use the simulator?

Thank you all,
PunchShadow

EricBoiseLGSVL, 2020-03-31:
@PunchShadow We have not tested with VNC Maybe [this](https://forum.unity.com/threads/trying-to-run-a-linux-standalone-on-aws-ec2-instance.425234/) will help?

What exactly are you trying to do?  Any idea @martins-mozeiko ?

martins-mozeiko, 2020-03-31:
If you are running X11 server under VNC server, then it won't work as applications won't have access to GPU rendering - VNC server creates virtual display that X11 uses.

You need to run VNC server uinder X11 servert. Basically install and run X11 same as you would do locally. Then run VNC server that will capture desktop rendered by real GPU.

PunchShadow, 2020-04-01:
@EricBoiseLGSVL  @martins-mozeiko thanks for your fast response,
I've @martins-mozeiko 's method to use [x11vnc](https://help.ubuntu.com/community/VNC/Servers) instead of the traditional VNC server, and it solves "Xlib: extension missing display" bugs

However, it still can't successfully show the simulator after I select the resolution.
Is x11vnc the solution that @martins-mozeiko method about ?

Btw, I want to run simulator in and from [ADE](https://ade-cli.readthedocs.io/en/latest/install.html) container

martins-mozeiko, 2020-04-01:
Yes, x11vnc should work fine. What is the error you are getting? The logfile should have more details.

PunchShadow, 2020-04-01:
Here's my logfile
Under the log are some suggestions

![image](https://user-images.githubusercontent.com/32506160/78150908-84b48580-746a-11ea-
9f03-5e783e7b5abe.png)

I think it's because I use `x11vnc -create` and not start X server first.
Therefore, it can't open DISPLAY successfully.
But I'm still find the way to solve it.

tleyden, 2020-04-11:
@PunchShadow I wasn't able to get good performance on any of the vnc servers from my OSX machine on my wifi network and so I switched to [NoMachine](https://www.nomachine.com/), which I can confirm works with the LGSVL simulator.  Good luck!

PunchShadow, 2020-04-12:
@tleyden Thank you very much!
I'll try it as soon as possible.

EricBoiseLGSVL, 2020-04-16:
@tleyden Did @PunchShadow suggestion solve this issue?  If so, please confirm and close issue

tleyden, 2020-04-16:
@EricBoiseLGSVL actually @PunchShadow opened the issue, I just noticed it and was giving a tip that worked for me.  It doesn't sound like an LGSVL issue though, so I would suggest to close it.

EricBoiseLGSVL, 2020-04-16:
@tleyden Whoops, thanks.  I flipped it :) Thanks @tleyden for suggestion, @PunchShadow did it solve your issue?

PunchShadow, 2020-04-27:
@EricBoiseLGSVL @tleyden  I've tried NoMachine, but still can't open it successfully.
I refer the setting of [NoMachine Setting on headless server](https://www.nomachine.com/AR10K00710) but showing the same bugs in #314 (I've already install the recommended packages).

Here's part of my log:

Desktop is 1024 x 768 @ 0 Hz
[Vulkan init] extensions: count=16
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
Vulkan detection: 0
No supported renderers found, exiting

(Filename: ./PlatformDependent/LinuxStandalone/main.cpp Line: 639)

EricBoiseLGSVL, 2020-04-28:
@PunchShadow Unity requires a GPU to run.  Headless will not work.

martins-mozeiko, 2020-04-28:
That is not quite accurate. Traditionally headless means there is no monitor attached. But as long as there is display server running, GPU present and nvidia driver installed - Unity will run fine. Regardless of where output goes (headless or no headless).

I don't know how exactly NoMachine works, but if it uses xvfb virtual framebuffer for it's display server, then it simply won't have access to GPU. My suggestion is to run actual, real X11 display server, so it can have full access to GPU. Then run any VNC solution that captures it's desktop.

I'd suggest first getting any GPU oriented application to work on your chosen solution (NoMachine) before trying to run simulator. Only after it works, then run simulator. So start running `glxgears -info`  and make sure Nvidia and GPU name appears in its output. Then install [Vulkan SDK](https://vulkan.lunarg.com/doc/view/1.1.101.0/linux/getting_started_ubuntu.html) and run `vkcube` - make sure it works. Once you get here, the simulator should run fine afterwards.

PunchShadow, 2020-04-28:
@EricBoiseLGSVL @martins-mozeiko I can run Nvidia-docker with nvidia-smi, so I think it's not the driver or GPU's problem. I'll try to operate another GPU oriented application first and figure out the more precise question. Thanks @martins-mozeiko again !

tleyden, 2020-09-13:
@PunchShadow I may have found out why NoMachine didn't work for you -- I just hit the same error.  The way I fixed it was to first login on the machine itself (I have physical access to the machine, it's not in the cloud), then connect from NoMachine.  I think there is some subtle difference in the way NoMachine works based on what @martins-mozeiko mentioned.

Also, it's worth mentioning that older versions of NoMachine seemed to force me to initially login on the machine itself (thus avoiding the error), but newer versions (6.11.2-3) can create the desktop session from scratch (thus hitting the error).

In summary..

**This didn't work**

* Boot ubuntu 18 machine
* Connect from OSX via NoMachine client
* Run `vkcube`

Error:

```
$ vkcube
vkcube: /build/vulkan-tools-1.2.148.0-1ubuntu18.04/cube/cube.c:3176: demo_init_vk: Assertion `!err' failed.
Aborted
```

If I try to run lgsvl simulator, I see the same errors as https://github.com/lgsvl/simulator/issues/708#issuecomment-619717408

**This worked**

* Boot ubuntu 18 machine
* Login to the physical machine, shows desktop
* Run `vkcube`
* Connect from OSX via NoMachine client
* See spinning `vkcube`, can run lgsvl simulator

EricBoiseLGSVL, 2020-09-15:
@tleyden Thanks for the info.  I hope this helps!

closed 2020-10-20 by EricBoiseLGSVL