lgsvl/simulator #709: Junction Lane's turn value are invalid for San Francisco Map

ntutangyun, 2020-04-03:
Hi team, 

I found that in the `base_map` file of san francisco map, all the lanes belonging to junctions have the same `turn` attribute to be 1, which is `NO TURN` according to Apollo's map specificiation. 
examples in Junction `J_0`: 

[image]

`lane_659` should be labeled as right turn and `lane_652` should be labeled as left turn.
But in the `base_map`, both have been labeled `NO TURN`...

Junction `J_0` is here: (around center of the below map)
[image]

Meanwhile, the `turn` attributes are correct for `Borregas Avenue` map. 

Please kindly check it out. Thanks.

ntutangyun, 2020-04-03:
Found another issue: 
San Francisco Map -> Junction 70 -> lane 1312: the first two points seems to be inverted...

[image]

Junction 70 is here (right bottom corner)
[image]

ntutangyun, 2020-04-03:
Here's a list of lanes within Junctions which do not but should have predecessor lanes:
* `lane_870` of junction `J_19`
* `lane_1035` of junction `J_34`
* `lane_1312` of junction `J_70`

Here's a list of lanes within Junctions which do not but should have successors lanes:
* `lane_940` and `lane_950` of junction `J_25`
* `lane_1066` of junction `J_38`
* `lane_1163` of junction `J_49`
* `lane_1333` of junction `J_72`

Please kindly check and fix also. thanks.

EricBoiseLGSVL, 2020-04-03:
@ntutangyun Thanks for finding these.  We will update when as soon as we are able.

luqiang21, 2020-04-07:
@ntutangyun Thank you for these detailed findings! I've checked our SF map. 
- For the list of lanes wich missing predecessor/successor lanes, I inspected each lane and found that their end points are not overlapping with desired endpoints of other lanes. I manually adjusted all of them, now I can see all these lanes have predecessor/successor lanes.
- For intersection lanes' turn types, we didn't annotate them at all. I annotated every turning lane for all 90+ intersections in SF map. 
- [This](https://drive.google.com/open?id=1u1Mkw6wyW2I0CNZu0TH-T3qR8wNOVFaE) is the new `bin` file for you. We will update map file in the repos soon. Please let me know if you found any lane has incorrect turn type or missing something.

Thanks.
BTW, what tools are you using to visualize the map?

ntutangyun, 2020-04-08:
@luqiang21 Thanks for your prompt help. 

> what tools are you using to visualize the map?

I made the visualizer myself. 
I've used Javascript protobuf tools to parse the binary files, and I used [three.js](https://threejs.org/) library to plot those points and arrows in browser.

> This is the new bin file for you. We will update map file in the repos soon. Please let me know if you found any lane has incorrect turn type or missing something.

Thanks again. Sure will follow up with this issue when i found anything.

ntutangyun, 2020-12-22:
Hi, I'm not sure if the `San Francisco` map is still under maintenance

But I found that some of the junction does not have all their junction lanes included. 

example: Junction `J_84` is supposed to have many lanes with id "lane_144x" as shown below:

[image]

This is what it has:
  'lane_1426', 'lane_1427',
  'lane_1428', 'lane_1429',
  'lane_1430', 'lane_1431',
  'lane_1432', 'lane_1433',
  'lane_1434', 'lane_1435',
  'lane_1436', 'lane_1437',
  'lane_1438', 'lane_1439',
  'lane_1440', 'lane_1441'

Please help check it out if possible. 

Thanks.

ntutangyun, 2020-12-22:
also, seems that the crosswalks are exported without ids.

EricBoiseLGSVL, 2020-12-22:
We are still working on SF.  We will look into it, thanks @ntutangyun