lgsvl/simulator #710: imu data not publishing

ajay1606, 2020-04-06:
Thanks to LGSVL for an updated version of the simulator package.
I have a few quick questions regarding simulator sensor data output to verify with the Autoware.

1. command rostopic list output following topics:
/client_count
/connected_clients
/imu_raw
/nmea_sentence
/odom
/points_raw
/rosout
/rosout_agg
/simulator/camera_node/image/compressed
But there are no any data publishing in the topic /imu_raw

2. Raw points data are getting at /points_raw topic, but the output rate of is less than 5hz.

Could you please help me to figure out how to get imu_raw data and points_raw at 10 hz frequency.

Kind regards,

EricBoiseLGSVL, 2020-04-06:
@hadiTab isn't this an issue with ROS?

hadiTab, 2020-04-07:
@ajay1606 what sensors do you have on the vehicle? Is it the default Autoware car?
Also what are the specs of the computer you are running it on?

Do you get any errors when you try to echo imu_raw?

ajay1606, 2020-04-07:
@EricBoiseLGSVL & @hadiTab Thanks for your quick response. Currently am running LGSVL with Autoware with following specifications

OS: Ubuntu 16.04 (32GB RAM, Graphics card Titan Xp)
ROS: kinetic

Simulation setup:
[image]
Vehicle and map setup:
[image]
Rosbridge connection status:
[image]
$rostopic list output:
[image]
$rostopic echo /poinst_raw output:
[image]
$rostopic echo /imu_raw output:
[image]
$rostopic hz /points_raw
[image]
$rostopic hz /imu_raw
[image]

Kind Regards
Ajay

EricBoiseLGSVL, 2020-04-07:
@ajay1606 Can you post player.log as well as any error msg from imu_raw please

ajay1606, 2020-04-08:
@EricBoiseLGSVL I could not find the player.log file in the  location: /.config/unity3d/LG Silicon Valley Lab/ 
Also, sometimes I observe very strangely /imu_raw publish and pause as shown in the shared [video](https://drive.google.com/open?id=1PUvsUuWL9sQnF07EZjB7xd4ubLs6QMfw)

hadiTab, 2020-04-08:
@ajay1606 the rosbridge for ros kinetic has a lot of performance issues. Ros melodic's rosbridge does a much better job, but you can't install it on your host machine because it requires ubuntu 18.04. You can run it through docker though. I would suggest testing it out on ros melodic through a docker container to see if that improves things.

For example if you have docker installed:
1. `docker pull ros:melodic-ros-core`
2. run container: `docker run -it --net host ros:melodic-ros-core`
3. inside container install rosbridge and run it:
```
apt update
apt install ros-melodic-rosbridge-suite
roslaunch rosbridge_server rosbridge_websocket.launch
```
4. listen to topics on your host machine and see if things improve

ajay1606, 2020-04-08:
@hadiTab , thanks for your efforts, I will check with ubuntu 18.04 and get back to you.

ajay1606, 2020-04-09:
@hadiTab actually I have setup ubuntu 18.04 with ROS melodic in a separate machine. But I am getting an error while starting the simulation environment. 

1. Test with lgsvlsimulator-linux64-2020.03-rc1 (release version)
[image]

2. lgsvlsimulator-linux64-2020.01 (release version)
[image]

Sorry for bothering you again, but could you please help me to fix this issue.

kind regards,

hadiTab, 2020-04-09:
It seems like you had an old simulator installation on that machine.

Close the simulator and delete the folder called "LG Silicon Valley Lab" located in ~/.config/unity3d.

Launching the simulator again now should redownload all the maps and vehicles and the error should be resolved.

ajay1606, 2020-04-09:
Excellent, billion thanks to you.

1. getting data imu_raw @ 90 hz
2. getting points_raw @10 hz

Really appreciate your help.

closed 2020-04-09 by ajay1606