lgsvl/simulator #711: San Francisco map  -  mismatch between Physical map in game and HD map file

ntutangyun, 2020-04-06:
Hi team,

I've found three mismatches between physical map of San Francisco in game and the HD map that Apollo reads.

***Stopping line mismatch at Junction `J_90`***
https://drive.google.com/file/d/18SjuZ5lzj9yE8Q0WZHbyWMMHbydL8drE/view?usp=sharing

***Misaligned right turning junction lane at junction `J_76`***
https://drive.google.com/file/d/1QQjcvetyHRYFGlbZGqTZX81K0vn-0aRg/view?usp=sharing
I'm not quite sure about this one, but there is certainly a lot of space on the right. 
The lane boundary is not matched with the physical curb.
Sometimes Apollo even stops due to the sharp routing path.

***invisible curb at Junction `J_63`***
https://drive.google.com/file/d/1iUTwtN9u1W_B6MzhqZvG1-a8Stw4JS9f/view?usp=sharing
Note that Apollo is not using camera or lidar to improve FPS. 
Thus what Apollo knows is only its ground truth location and a HD map. 
But there seems to be a tiny mismatch in the map, which resulted in Apollo hitting the curb. 

Please kindly check them out. Thanks.

EricBoiseLGSVL, 2020-04-06:
@luqiang21 can you look at this please.

EricBoiseLGSVL, 2020-04-07:
@ntutangyun Thank you for letting us know about this.  We will add this to the backlog of upcoming fixes