lgsvl/simulator #712: out of order odometry msgs

alexsimon1, 2020-04-06:
Hello,

the odometry msgs in my rosbag are not in order.
I mean the position in the world not the timestamps.
the timestamps are correct.

completely random the pose of the odom msg jumps to a coordinate in the Trajectory the Vehicle took in the past. 

Is this a known bug?

Is there a solution ?

EricBoiseLGSVL, 2020-04-06:
@daviduhm do you know what could cause this issue?

alexsimon1, 2020-04-06:
i should note that i changed the header timestamps of the rosbag with the timestamps of the msgs. 
With 1.6 from 

http://wiki.ros.org/rosbag/Cookbook

and the simulator ran in Linux.

alexsimon1, 2020-04-08:
is there maybe some sort of buffer for the odometry msg in the lgsvl simulator ?

daviduhm, 2020-04-15:
Hi, I haven't seen this issue before and I believe [GPS odometry sensor](https://www.lgsvlsimulator.com/docs/sensor-json-options/#gps-odometry) shouldn't be like this. 

@alexsimon1 I'll try to reproduce it. Could you elaborate on how you are running this in more detail? (i.e., which version/commit of simulator are you using, what sensors do you have in your sensor json, are you running any AV stacks such as apollo or autoware with simulator, how are you checking the trajectory, and etc).

alexsimon1, 2020-04-15:
Hello, i‘m sorry it was a problem with my scripts. Im closing the issue.

closed 2020-04-15 by alexsimon1