lgsvl/simulator #713: Control vehicle from ROS , WINDOWS

Erdemhan, 2020-04-07:
Hi,
I am trying to control a vehicle from ros. Actually i want to:
Get camera and lidar datas from vehicle
send to my image processing script (with ros or anyway)
get instructions from my script for control the car(with ros or anyway)

I know that running simulator in windows is better . I am using ROS1 on windows i found sth about apollo and autoware but only infos i can find about these are about linux.
Can i use them with windows if yes how?

Thanks and sorry for my english if there are mistakes.

Erdemhan, 2020-04-07:
I think its not supported for windows now using it with linux

closed 2020-04-07 by Erdemhan

martins-mozeiko, 2020-04-08:
Yes, Apollo and Autoware works only on Linux
But if you have ROS1 environment on Windows, then you can communicate to Simulator just fine without them. All you need to install is [rosbridge_suite](https://github.com/RobotWebTools/rosbridge_suite) package and run rosbridge_websocket.launch. After that you'll be able to talk to simulator by using regular ROS messages - both for receiving sensor messages and sending messages to simulator.

Erdemhan, 2020-04-08:
> Yes, Apollo and Autoware works only on Linux
> But if you have ROS1 environment on Windows, then you can communicate to Simulator just fine without them. All you need to install is [rosbridge_suite](https://github.com/RobotWebTools/rosbridge_suite) package and run rosbridge_websocket.launch. After that you'll be able to talk to simulator by using regular ROS messages - both for receiving sensor messages and sending messages to simulator. Even on commandline - with `rostopic`.

Thanks! I just tried it with linux but there is a problem. When i try to get color camera images from apollo and save it to a jpeg file everything works fine but when i try to do it same with rosbridge it gives error 
`cv2.imwrite('a.jpeg',cv2.CV_LOAD_IMAGE_COLOR)
AttributeError: 'module' object has no attribute 'CV_LOAD_IMAGE_COLOR`

if i change the cv2.CV_LOAD_IMAGE_COLOR to 1 or IMREAD_COLOR it gives a 1x4 black image
but same code is working on apollo

and here is my code to get and save camera image:
` img_ = rospy.wait_for_message('simulator/main_camera', CompressedImage)

    img_np_arr = np.fromstring(img_.data, np.uint8)
    encoded_img = cv2.imdecode(img_np_arr,cv2.IMREAD_COLOR)
    cv2.imwrite('a.jpeg',cv2.cv2.CV_LOAD_IMAGE_COLOR)`