lgsvl/simulator #715: ROS bridge black camera image

Erdemhan, 2020-04-08:
Hi,
When i try to get color camera images from apollo and save it to a jpeg file everything works fine but when i try to do it same with rosbridge it gives error:
`cv2.imwrite('a.jpeg',cv2.CV_LOAD_IMAGE_COLOR)
AttributeError: 'module' object has no attribute 'CV_LOAD_IMAGE_COLOR`

if i change the cv2.CV_LOAD_IMAGE_COLOR to 1 or IMREAD_COLOR it gives a 1x4 black image
but same code is working on apollo
and when i make rostopic echo everything looks similar to apollo

and here is my code to get and save camera image:
` img_ = rospy.wait_for_message('simulator/main_camera', CompressedImage)

    img_np_arr = np.fromstring(img_.data, np.uint8)
    encoded_img = cv2.imdecode(img_np_arr,cv2.IMREAD_COLOR)
    cv2.imwrite('a.jpeg',cv2.cv2.CV_LOAD_IMAGE_COLOR)`

Camera is working on simulator and here is my camera setting
```

{
    "type": "Color Camera",
    "name": "Main Camera",
    "params": {
      "Width": 1920,
      "Height": 1080,
      "Frequency": 15,
      "JpegQuality": 75,
      "FieldOfView": 50,
      "MinDistance": 0.1,
      "MaxDistance": 1000,
      "Topic": "/simulator/main_camera",
      "Frame": "camera",
      "Distorted": true,
      "DistortionParameters": [
        -0.25349, 0.11868, 0, 0
      ]
    },
    "transform": {
      "x": 0,
      "y": 1.7,
      "z": -0.2,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
}
```

Erdemhan, 2020-04-09:
Found the exception 
it is in code:
`cv2.imwrite('a.jpeg',cv2.CV_LOAD_IMAGE_COLOR)``
Because of differen opencv verisons (maybe because of docker idk)
its working in apollo but not working in my own ros workspace
solution is:
`cv2.imwrite('a.jpeg')`

closed 2020-04-09 by Erdemhan

enginlien, 2021-08-06:
@Erdemhan 

Hi
How to access camera images?
Thanks