lgsvl/simulator #716: Folder name of release 2020.03-rc1 doesn't contain `-rc1`

left4taco, 2020-04-09:
Hi there,

We have set up CI to create volume of lgsvl simulator used by docker container. However, the job failed for release 2020.03-rc1. https://gitlab.com/ApexAI/ade-lgsvl/-/jobs/504659281

In the error trace it shows that the folder name within the zip file is `lgsvlsimulator-linux64-2020.03`. However, the job is looking for a folder called `lgsvlsimulator-linux64-2020.03-rc1`.

The behavior was correct in release  2020.01-rc1, though. https://gitlab.com/ApexAI/ade-lgsvl/-/jobs/429907461

EricBoiseLGSVL, 2020-04-09:
@left4taco we will look into this, thanks

eagafonovlge, 2020-04-09:
@left4taco Thanks for the reporting!
We have updated zip files, folder name matches to the archive name.
Please restart CI pipeline to verify.

closed 2020-04-09 by eagafonovlge