lgsvl/simulator #717: vehicle_cmd  Vehicle just braking

Erdemhan, 2020-04-09:
I am using ROS and a LGSVL car with ROS bridge i am able to send and get datas but the problem is
when i send acceleration data to vehicle_cmd with:

 `rostopic pub -r 10 /vehicle_cmd autoware_msgs/VehicleCmd '{ctrl_cmd:{linear_acceleration: 1.5} , ctrl_cmd:{steering_angle: 5.0}}'`

doesn't give any erros and:
Vehice is starting to brake(brake lights turning on) and
AD Car Control says:
 ControlType AutowareAi
 AD Accel Input: -1.123123E-08 (Always changing)
 AD Steer Input 0

tried to send linear_acceleration both -1.5 and 1.5 and different values nothing changed.
How can i control the car with ros?

Not: I am not using autoware just my own ros and rosbridge

EricBoiseLGSVL, 2020-04-09:
@Erdemhan do you have a VehicleControlSensor in your vehicle setup?  Did you modify it to accept your ROS msg types?

Erdemhan, 2020-04-09:
> @Erdemhan do you have a VehicleControlSensor in your vehicle setup? Did you modify it to accept your ROS msg types?

I have :
```
{
    "type": "Vehicle Control",
    "name": "AD Car Control",
    "params": {
      "Topic": "/vehicle_cmd"
    }
}
```
in my vehicle , i can see it in rostopic list. I am trying to send ctrl_cmd.
I didnt modify it. do i have ? if yes how please.
i tried to send a message like 
`'{ctrl_cmd:{linear_acceleration: 1.5} , ctrl_cmd:{steering_angle: 5.0}}'`
from terminal when i send it vehicle starts to brake and
```
AD Car Control says:
ControlType AutowareAi
AD Accel Input: -1.123123E-08 (Always changing)
AD Steer Input 0
```
when i stop sending messages braking stops and parameters turns to:
```
ControlType AutowareAi
AD Accel Input: 0
AD Steer Input 0
```
as expected

because of cant use vehicle_cmd  i tried  to control vehicle from my keyboard with a python script but i need to know speed info for set throttle brake etc. and this time i cant get data from canbus and opened an issiue #718 about that.

EricBoiseLGSVL, 2020-04-13:
@hadiTab Do you know what the issue could be?

hadiTab, 2020-04-17:
Can you try populating the `twist_cmd` portion of the message only?
I believe that is the only part being populated by Autoware AI ans the simulator may only be listening to that.

I believe the braking means that the `twist_cmd` is getting set to all 0s.

Erdemhan, 2020-04-17:
@hadiTab @EricBoiseLGSVL  **This:**

> Can you try populating the `twist_cmd` portion of the message only?
> I believe that is the only part being populated by Autoware AI ans the simulator may only be listening to that.
> 
> I believe the braking means that the `twist_cmd` is getting set to all 0s.

 solved my problem it worked with `twist_cmd` thanks for helping.

closed 2020-04-17 by Erdemhan

amirtakouh, 2020-07-27:
@hadiTab @EricBoiseLGSVL 
Could you please clarify the twist_cmd? It can be used for steering and accelerating the vehicle, but how about the other functionalities? for example, changing gear?
When trying to control the simulated vehicle by vehicle_cmd autoware_msgs/VehicleCmd, as you pointed out, it seems only the twist_cmd is responded to. Could you please clarify how the vehicle could be controlled over this topic with it's full functionality?

EricBoiseLGSVL, 2020-07-28:
@amirtakouh This issue is answered and closed.  To better issues easier to search for other users, please start a new issue.  Thanks