lgsvl/simulator #719: How to modify the speed of pedestrian?

CurlyWei, 2020-04-10:
Hi, the current version of "Python API", it doesn`t modify the speed of pedestrian. Is there any way to do it? Will it be released in the future?

Thanks

EricBoiseLGSVL, 2020-04-13:
@CurlyWei The API cannot modify the speed of pedestrians.  If you want to change it, I would expose the LinearSpeed of the Agent in PedestrianController.cs and make a new Python command to alter.  You could also edit the create-agent to set a new speed on spawn.  What is your use case?  Can you explain more and maybe we can include in a future feature?

CurlyWei, 2020-04-29:
Sorry for the late reply,I hava a project which pedestrian test scenarios.This content describes crash with pedestrian.
Please reference NHTSA:A Framework for Automated Driving System Testable Cases and Scenarios
In order to simulate behavior of pedestrian,I wanna modify the speed of pedestrians.
And thanks for your reply.

EricBoiseLGSVL, 2020-04-29:
@CurlyWei no worries.  I'll see how much work this feature will be and try to add to our roadmap.  I'll post with more details soon

EricBoiseLGSVL, 2020-10-20:
In latest release

closed 2020-10-20 by EricBoiseLGSVL

CurlyWei, 2020-10-29:
Thanks for responding so quickly to my request.
It`s very effective.