lgsvl/simulator #721: How to control vehicle using /vehicle_cmd(with ctrl_cmd, without twist_cmd)

atinfinity, 2020-04-12:
I tried to control vehicle using `/vehicle_cmd`(with ctrl_cmd, without twist_cmd).   
The `autoware_msgs/VehicleCmd` has `twist_cmd` and `ctrl_cmd`.
https://gitlab.com/autowarefoundation/autoware.ai/messages/-/blob/1.13.0/autoware_msgs/msg/VehicleCmd.msg#L8-9

So, I refered to [BorregasAve/my_launch/my_motion_planning.launch](https://github.com/lgsvl/autoware-data/blob/master/BorregasAve/my_launch/my_motion_planning.launch).
Therefore, I used `pure_pursuit` and `twist_filter` for motion_planning.  These packages publish ControlCommand. And, I set gear `/vehicle_cmd` of to 64.
https://github.com/lgsvl/simulator/blob/2020.01/Assets/Scripts/Bridge/Ros/RosConversions.cs#L514-L515

I have some questions.

### Range of ControlCommand parameters
The `ControlCommand` has the following parameters.
<https://gitlab.com/autowarefoundation/autoware.ai/messages/-/blob/1.13.0/autoware_msgs/msg/ControlCommand.msg>

- linear_velocity
- linear_acceleration
- steering_angle

And, LGSVL simulator uses `linear_acceleration` and `steering_angle` when gear is 64.
https://github.com/lgsvl/simulator/blob/2020.01/Assets/Scripts/Bridge/Ros/RosConversions.cs

I could not find the range of these parameters in LGSVL simulator documentation.
I found related issue <https://github.com/lgsvl/simulator/issues/472#issuecomment-550570394>.

### How to convert parameter
The `pure_pursuit` publishes ControlCommand with `linear_acceleration`.
And, a unit of this `linear_acceleration` is `m/s^2`. 
https://gitlab.com/autowarefoundation/autoware.ai/messages/-/blob/1.13.0/autoware_msgs/msg/ControlCommand.msg#L2

But, I guess that LGSVL simulator assumes different unit.
So, I could not understand to convert parameter for LGSVL simulator.

EricBoiseLGSVL, 2020-04-13:
@atinfinity We are currently working on updating this feature.

atinfinity, 2020-04-14:
@EricBoiseLGSVL 

> We are currently working on updating this feature.

Thank you for your reply. I have two questions.

- Could you tell us the limiation of this feature at this time?
- Do you have a milestone about this feature?

EricBoiseLGSVL, 2020-04-15:
@atinfinity When we implemented autoware support, we couldn't get certain features to work and had to make our own conversions.
We should have more information within the next couple of months

atinfinity, 2020-04-17:
@EricBoiseLGSVL Thank you for your answer. I understood current status.

tajinder96, 2020-10-06:
Hi wanted to check if there have been any updates so far. Is it now possible to directly use acceleration setpoints (m/s2) or is acceleration still only possible through adjusting the throttle percentage?

EricBoiseLGSVL, 2020-10-06:
@tajinder96 Yes, currently it is through throttle.  Right @hadiTab ?
Yes, you will need to edit the controller script or make a new one the uses IVehicleDynamics.  You could then ignore the inputs and get the acceleration setpoints and apply torque.  Be sure you build a new vehicle bundle and a new simulator binary with this new controller otherwise simulator won't recognize the class.

tajinder96, 2020-10-12:
@EricBoiseLGSVL Thank you! :)

closed 2021-07-16 by EricBoiseLGSVL