lgsvl/simulator #722: Can I adjust the frequency of IMU and get "sensor_msgs/Image" rather than "sensor_msgs/CompressedImage"?

chen1qy, 2020-04-14:
I need camera data in type of "sensor_msgs/Image" , but I only can find "sensor_msgs/CompressedImage". Is there some way to get "sensor_msgs/Image" message? And the IMU sensor output at a fixed rate of 100 Hz. Can I adjust the frequency to 200Hz?

EricBoiseLGSVL, 2020-04-16:
@chen1qy Yes, you can create a custom camera sensor or edit the existing sensor code to send the data that you need.  You will also need to add a custom msg type.  Be aware that a non compressed image with slow down data transfer.  Same with IMU sensor.  You can alter the frequency in the sensor code.

chen1qy, 2020-04-17:
If I use the standalone executable, where can I edit or create sensor code? the JSON configuration?

martins-mozeiko, 2020-04-17:
I think it would be much easier to use ROS [image_transport](http://wiki.ros.org/image_transport#Usage) package to receive compressed image and republish uncompressed image. It will be higher performance to do this way, as simulator would need to send less data to ROS.

As for rising IMU frequency - you cannot go above 100Hz, because IMU and other physics related sensors are tied to physics simulation frequency in Unity. Technically you can rise physics simulation frequency for whole proejct, but  that will require extra fixes for car movement as now simulation will happen a bit differently.

EricBoiseLGSVL, 2020-04-17:
@chen1qy No, if you are not editing source code, you cannot change these parameters.  Thanks @martins-mozeiko for explaining better options

chen1qy, 2020-04-17:
Thanks for your help.

closed 2020-04-17 by chen1qy