lgsvl/simulator #723: How to use Unity scene for LGSVL simulation

bage131, 2020-04-15:
I imported "QA Modular Parking 1.01.unitypackage"  to Unity3D,Then I created AssetBundles with Demo scene. Its name is parkingv1.There was an error "Failed to add map: You must specify a valid AssetBundle" when I added map with parkingv1.But It was OK when I added new map with environment_CubeTown.Could you tell me how to solve this problem? Are there any more details about how to use the scene in Unity for LGSVL simulation? Thank you.
[image]

Erdemhan, 2020-04-15:
I dont know what's inside your scene but if you want to do something like building a new map with unity and adding it to simulator you should build it with unity from simulator tab but there are more details about it. You can find all steps at #699

EricBoiseLGSVL, 2020-04-16:
@bage131 yes, @Erdemhan is correct.  If you correctly created a map, it will be named "environment_YourMap".  YourMap will match folder in External/Environments and also match YourMap.scene file in that folder.  Then Simulator -> Build... will list the map and create an Asset Bundle, not a unity package.  unity package is for creating assets to distribute internally or to sell on Unity Asset Store.

We just released yesterday a tutorial video to help explain environment creation! [Here](https://youtu.be/S0w2ahhEPbE)

bage131, 2020-04-23:
Thank you very much.But I have a new problem.I have created a  parking map in the Unity.The car is far from the ground when I  run the simulator.As shown in the picture.How to set the position on the map.Thank you.
[image]

Erdemhan, 2020-04-23:
> Thank you very much.But I have a new problem.I have created a parking map in the Unity.The car is far from the ground when I run the simulator.As shown in the picture.How to set the position on the map.Thank you.
> [image]

You should set SpawnInfo's position.

closed 2020-04-29 by EricBoiseLGSVL

bage131, 2020-06-02:
I got it.Thank you very much.