lgsvl/simulator #724: load assetbundle error

yujie181152, 2020-04-16:
hi,i have got an error when i run python example script,the error info i posted like this:

`SerializationException: Property 'assetGuid' not found on type 'Simulator.Manifest'.
YamlDotNet.Serialization.TypeInspectors.TypeInspectorSkeleton.GetProperty (System.Type type, System.Object container, System.String name, System.Boolean ignoreUnmatched) (at <00b2fecf29ac47848149aa4ed801d87f>:0)
YamlDotNet.Serialization.NodeDeserializers.ObjectNodeDeserializer.YamlDotNet.Serialization.INodeDeserializer.Deserialize (YamlDotNet.Core.IParser parser, System.Type expectedType, System.Func`3[T1,T2,TResult] nestedObjectDeserializer, System.Object& value) (at <00b2fecf29ac47848149aa4ed801d87f>:0)
YamlDotNet.Serialization.ValueDeserializers.NodeValueDeserializer.DeserializeValue (YamlDotNet.Core.IParser parser, System.Type expectedType, YamlDotNet.Serialization.Utilities.SerializerState state, YamlDotNet.Serialization.IValueDeserializer nestedObjectDeserializer) (at <00b2fecf29ac47848149aa4ed801d87f>:0)
Rethrow as YamlException: (Line: 1, Col: 1, Idx: 0) - (Line: 1, Col: 1, Idx: 0): Excep`

the map and vehicle i used was downloaded from lgsvlsimulator.com,web console shows assetbundle is downloaded and vaild,like this, what i missed,is there anyone who know this problem?thank you very much  

[image]

mingw12, 2020-04-16:
code  is not match fit . see [](url)https://github.com/lgsvl/simulator/issues/30#issuecomment-437495513
reclone simulator , add and bulid   Assets to   assetbundle.

EricBoiseLGSVL, 2020-04-16:
@yujie181152 please delete simulator persistent data folder located at:
- Windows C:\Users\[username]\AppData\LocalLow\LG Silicon Valley Lab\LGSVL Simulator
- Linux /home/[username]/.config/unity3d/LG Silicon Valley Lab/LGSVL Simulator

Then run binary and all correct bundles will download.  AppData folder needs you to enable hidden folders on Windows.

yujie181152, 2020-04-18:
> @yujie181152 please delete simulator persistent data folder located at:
>
> * Windows C:\Users[username]\AppData\LocalLow\LG Silicon Valley Lab\LGSVL Simulator
> * Linux [...]

thanks! how to run binary?i don't understand this command.and i wanne know that if i can use old simulator source release to load new assetbundle which from website downloaded ,for example,i use 2019.10f1 version to load the latest assetbundle ,is it ok?

yujie181152, 2020-04-20:
thanks! how to run binary?i don't understand this command.and i wanne know that if i can use old released simulator source  to load new assetbundle which from website downloaded ,for example,i use 2019.10f1 version to load the latest assetbundle ,is it ok?

EricBoiseLGSVL, 2020-10-20:
You need to use new bundles with new releases.  Old simulator will not work with new bundles

closed 2020-10-20 by EricBoiseLGSVL