lgsvl/simulator #725: Migration Guide from 2020.01

joshwapohlmann, 2020-04-16:
Please provide a migration guide in the docs explaining how to migrate.
* Ego Vehicles
* Sensors
* Environments
* Controllables

As well as what's new (e.g. DistributionObject|Type|...)
Thanks :+1:

joshwapohlmann, 2020-04-16:
Notes
## Vehicles
**Changes**
* Change script from `VehicleDynamics` to `VehicleSMI`
* Cluster functionality requires now, that vehicles also have a `DistributedObject` and `DistributedRigidbody` script component attached for proper simulation synchronization policies.

**New**
* (opt.) Create a custom Vehicle Dynamic component by implementing the `IVehicleDynamics` interface.

## Sensors
**Changes**
None required

**New**
* (opt.) The `SensorBase` interface now defines a `DistributionType` which is set to `SensorDistributionType.DoNotDistribute` by default. Change this if required. See [the docs](https://www.lgsvlsimulator.com/docs/sensors-distribution/) for details.

## Environment
**Changes**
None

**New**
None

## Controllables
**no experience**

EricBoiseLGSVL, 2020-04-16:
@joshwapohlmann Migration guide is a good idea, thanks.
- Yes, we had many requests for users to be able to create their own dynamics models so this was a big change.  We are currently updating tutorial docs and videos to help migration.
- Distributed is somewhat in early we plan on documenting this more also.

closed 2021-07-16 by EricBoiseLGSVL