lgsvl/simulator #726: Running the simulator issue

Martin20, 2020-04-16:
I use the simulator for windows version. When I wan to start a simulation I receive the error below:

Invalid: (Line: 1, Col: 1, Idx: 0) - (Line: 1, Col: 1, Idx: 0): Exception during deserialization

EricBoiseLGSVL, 2020-04-16:
@Martin20 This is probably an issue with Yaml.  Can you please post more information?
Please post Player.log in full
What GPU are you using?
Are you running binary or from editor?
Is this an error in webUI or in the application?  
What environment are you loading?
More details will help us to solve the problem as well as others that might have the same issue.

Martin20, 2020-04-17:
Where is the palayer.log? I could not see it.
the GPU is Nvidia with driver:  Driver Version: 440.33.01
I run the binary.
it is in WebUI.
What do you mean by the environment? I run it on windows.

rongguodong, 2020-04-17:
It's at %APPDATA%\..\LocalLow\LG Silicon Valley Lab\LGSVL Simulator\Player.log

You can find it in our FAQ: https://www.lgsvlsimulator.com/docs/faq/#where-are-unity-log-files-located

EricBoiseLGSVL, 2020-04-17:
@Martin20 environment is the map that you are loading.  What is the GPU model and ram?  Also what version of simulator are you using?

Martin20, 2020-04-17:
@rongguodong 

the Player.log is here:

Mono path[0] = 'C:/lgsvlsimulator-windows64-2020.03-rc1/lgsvlsimulator-windows64-2020.03-rc1/simulator_Data/Managed'
Mono config path = 'C:/lgsvlsimulator-windows64-2020.03-rc1/lgsvlsimulator-windows64-2020.03-rc1/MonoBleedingEdge/etc'
Loading player data from C:/lgsvlsimulator-windows64-2020.03-rc1/lgsvlsimulator-windows64-2020.03-rc1/simulator_Data/data.unity3d
Initialize engine version: 2019.3.3f1 (7ceaae5f7503)
[Subsystems] Discovering subsystems at path C:/lgsvlsimulator-windows64-2020.03-rc1/lgsvlsimulator-windows64-2020.03-rc1/simulator_Data/UnitySubsystems
GfxDevice: creating device client; threaded=1
Direct3D:
    Version:  Direct3D 11.0 [level 11.1]
    Renderer: Intel(R) HD Graphics 4600 (ID=0x412)
    Vendor:   
    VRAM:     2048 MB
    Driver:   20.19.15.4835
Begin MonoManager ReloadAssembly
- Completed reload, in  0.137 seconds
WARNING: Shader Unsupported: 'HDRP/Lit' - All passes removed
WARNING: Shader Did you use #pragma only_renderers and omit this platform?
WARNING: Shader Unsupported: 'Shader Graphs/EnvironmentSimulation' - All passes removed
WARNING: Shader Did you use #pragma only_renderers and omit this platform?
WARNING: Shader Unsupported: 'Shader Graphs/CloudGenerator' - All passes removed
WARNING: Shader Did you use #pragma only_renderers and omit this platform?
D3D11 device created for Microsoft Media Foundation video decoding.
<RI> Initializing input.

New input system (experimental) initialized
XInput1_3.dll not found. Trying XInput9_1_0.dll instead...
<RI> Input initialized.

<RI> Initialized touch support.

UnloadTime: 0.953500 ms
WARNING: Shader Unsupported: 'Hidden/PostProcessing/FinalPass' - Pass '' has no vertex shader
WARNING: Shader Unsupported: 'Hidden/PostProcessing/FinalPass' - Pass '' has no vertex shader
WARNING: Shader Unsupported: 'Hidden/PostProcessing/Uber' - Pass '' has no vertex shader
WARNING: Shader Unsupported: 'Hidden/PostProcessing/Uber' - Pass '' has no vertex shader
Loading CheckControlSensor 
(Filename: C:\buildslave\unity\build\Runtime/Export/Debug/Debug.bindings.h Line: 35)

Fallback handler could not load library C:/lgsvlsimulator-windows64-2020.03-rc1/lgsvlsimulator-windows64-2020.03-rc1/simulator_Data/Mono/data-0000025C14152940.dll
Loading VelodyneLidarSensor 
(Filename: C:\buildslave\unity\build\Runtime/Export/Debug/Debug.bindings.h Line: 35)

Fallback handler could not load library C:/lgsvlsimulator-windows64-2020.03-rc1/lgsvlsimulator-windows64-2020.03-rc1/simulator_Data/Mono/data-0000025C111CBBF0.dll
Fallback handler could not load library C:/lgsvlsimulator-windows64-2020.03-rc1/lgsvlsimulator-windows64-2020.03-rc1/simulator_Data/Mono/data-0000025C14136D10.dll
WARNING: Shader Unsupported: 'HDRP/Lit' - All passes removed
WARNING: Shader Did you use #pragma only_renderers and omit this platform?
Current Database Version: 3 
(Filename: C:\buildslave\unity\build\Runtime/Export/Debug/Debug.bindings.h Line: 35)

Final Database Version: 3 
(Filename: C:\buildslave\unity\build\Runtime/Export/Debug/Debug.bindings.h Line: 35)

Setting up 4 worker threads for Enlighten.
  Thread -> id: 2e0c -> priority: 1 
  Thread -> id: 8dc -> priority: 1 
  Thread -> id: 2790 -> priority: 1 
  Thread -> id: 2680 -> priority: 1 
D3D shader create error for pixel shader [0x80070057]
ShaderProgram is unsupported, but because jobified rendering is enabled the ShaderProgram can not be removed. 
(Filename:  Line: 445)

ShaderProgram is unsupported, but because jobified rendering is enabled the ShaderProgram can not be removed. 
(Filename:  Line: 445)

ShaderProgram is unsupported, but because jobified rendering is enabled the ShaderProgram can not be removed. 
(Filename:  Line: 445)

[... 468 lines cut ...]
(Filename:  Line: 445)

ShaderProgram is unsupported, but because jobified rendering is enabled the ShaderProgram can not be removed. 
(Filename:  Line: 445)

ShaderProgram is unsupported, but because jobified rendering is enabled the ShaderProgram can not be removed. 
(Filename:  Line: 445)

ShaderProgram is unsupported, but because jobified rendering is enabled the ShaderProgram can not be removed. 
(Filename:  Line: 445)

ShaderProgram is unsupported, but because jobified rendering is enabled the ShaderProgram can not be removed. 
(Filename:  Line: 445)

ShaderProgram is unsupported, but because jobified rendering is enabled the ShaderProgram can not be removed. 
(Filename:  Line: 445)

ShaderProgram is unsupported, but because jobified rendering is enabled the ShaderProgram can not be removed. 
(Filename:  Line: 445)

ShaderProgram is unsupported, but because jobified rendering is enabled the ShaderProgram can not be removed. 
(Filename:  Line: 445)

ShaderProgram is unsupported, but because jobified rendering is enabled the ShaderProgram can not be removed. 
(Filename:  Line: 445)

Martin20, 2020-04-17:
@EricBoiseLGSVL 
I load Cube Town and the model of GPU and its specifications are as below:

 VGA compatible controller: NVIDIA Corporation GP102 [TITAN Xp] (rev a1) (prog-if 00 [VGA controller])
        Subsystem: NVIDIA Corporation Device 11df
        Flags: bus master, fast devsel, latency 0, IRQ 84
        Memory at f9000000 (32-bit, non-prefetchable) [size=16M]
        Memory at e0000000 (64-bit, prefetchable) [size=256M]
        Memory at f0000000 (64-bit, prefetchable) [size=32M]
        I/O ports at f000 [size=128]
        [virtual] Expansion ROM at fa000000 [disabled] [size=512K]
        Capabilities: <access denied>
        Kernel driver in use: nvidia
        Kernel modules: nvidiafb, nouveau, nvidia_440, nvidia_440_drm

EricBoiseLGSVL, 2020-04-18:
@Martin20 Thanks for the info it helps.
"ShaderProgram is unsupported, but because jobified rendering is enabled the ShaderProgram can not be removed."
looks like Simulator application might be using an integrated GPU instead of the Titan.  Can you try opening NVIDIA Control panel / Manage 3D Settings / Program settings and Select preferred Graphics processor for the application.

Martin20, 2020-04-18:
However, older versions work (2019.10).

yujie181152, 2020-04-20:
i got the same issue with you ,old vesion of unity can not load the latest assetbundle on the website...wating for the right solution,

EricBoiseLGSVL, 2020-04-20:
@Martin20 It says from your player log "Intel(R) HD Graphics 4600 (ID=0x412)" with "2048 MB" were you able to force the correct GPU?

@yujie181152 You will need to update the asset bundles for the new version of Unity and Simulator.  Either re-download or re-create local ones

Martin20, 2020-04-23:
@EricBoiseLGSVL 
No, it is not the correct GPU. Mine is an Nvidia.
How can I fix that?

lemketron, 2020-04-23:
> @EricBoiseLGSVL
> No, it is not the correct GPU. Mine is an Nvidia.
> How can I fix that?

You’re running LGSVL Simulator on Windows, right?

Did you try opening the NVIDIA Control panel, going to “Manage 3D Settings“ and “Program settings“ and try selecting the preferred graphics processor for the application?

Martin20, 2020-04-23:
I run the LGSVL on windows and remotely Apollo 5.0 on ubuntu. You mean change the settings on Windows or Ubuntu? LGSVL on windows does not use GPU because there is no GPU on my windows. I just have it on ubuntu where the Apollo is run.

lemketron, 2020-04-23:
If you don’t have a discrete GPU on the Windows machine then you should consider running LGSVL Simulator on the Ubuntu system with the Titan GPU.

https://www.lgsvlsimulator.com/docs/getting-started/#getting-started

Martin20, 2020-04-23:
@lemketron 
How an old version is working on the same machine?

lemketron, 2020-04-23:
> @EricBoiseLGSVL
> I load Cube Town and the model of GPU and its specifications are as below:
> 
> VGA compatible controller: NVIDIA Corporation GP102 [TITAN Xp] (rev a1) (prog-if 00 [VGA controller])
> Subsystem: NVIDIA Corporation Device 11df
> Flags: bus master, fast devsel, latency 0, IRQ 84
> Memory at f9000000 (32-bit, non-prefetchable) [size=16M]
> Memory at e0000000 (64-bit, prefetchable) [size=256M]
> Memory at f0000000 (64-bit, prefetchable) [size=32M]
> I/O ports at f000 [size=128]
> [virtual] Expansion ROM at fa000000 [disabled] [size=512K]
> Capabilities: 
> Kernel driver in use: nvidia
> Kernel modules: nvidiafb, nouveau, nvidia_440, nvidia_440_drm

This was confusing, because I think you’re saying that this Titan Xp is on Ubuntu (where you are running Apollo, not the simulator). Instead it seems you are running the Simulator on a Windows machine with only Intel graphics.

That may have (barely) worked for older simulator version (before HDRP), but the shaders in newer HDRP versions of the simulator have different GPU requirements, as explained by your log file here: 

Initialize engine version: 2019.3.3f1 (7ceaae5f7503)
[Subsystems] Discovering subsystems at path C:/lgsvlsimulator-windows64-2020.03-rc1/lgsvlsimulator-windows64-2020.03-rc1/simulator_Data/UnitySubsystems
GfxDevice: creating device client; threaded=1
Direct3D:
    Version:  Direct3D 11.0 [level 11.1]
    Renderer: Intel(R) HD Graphics 4600 (ID=0x412)
    Vendor:   
    VRAM:     2048 MB
    Driver:   20.19.15.4835
Begin MonoManager ReloadAssembly
- Completed reload, in  0.137 seconds
WARNING: Shader Unsupported: 'HDRP/Lit' - All passes removed
WARNING: Shader Did you use #pragma only_renderers and omit this platform?
WARNING: Shader Unsupported: 'Shader Graphs/EnvironmentSimulation' - All passes removed
WARNING: Shader Did you use #pragma only_renderers and omit this platform?
WARNING: Shader Unsupported: 'Shader Graphs/CloudGenerator' - All passes removed
WARNING: Shader Did you use #pragma only_renderers and omit this platform?

You will either need to upgrade to Nvidia graphics on the Windows machine or try running the simulator on the Ubuntu machine.

closed 2020-04-29 by EricBoiseLGSVL