lgsvl/simulator #727: traffic light detection does not work in apollo

Martin20, 2020-04-17:
Hello all,

When I run the simulator with Apollo 5.0, the traffic light detection is not working. Although I make it on in the Dreamview, after a while it becomes off. Even it is on, it cannot detect the traffic light. The vehicle just stop at the intersection with traffic light.

ntutangyun, 2020-04-17:
Hi @Martin20 ,

There are several things you may check
1. Have you included camera sensor in the vehicle config from simulator's website interface
2. Is simulator is publishing the camera data to CyberRT in `cyber_monitor`
3. Start traffic light component in a separate terminal with `cyber_launch`, see if anything goes wrong.

EricBoiseLGSVL, 2020-04-17:
@Martin20 Yes you need this sensor
{
    "type": "Color Camera",
    "name": "Telephoto Camera",
    "params": {
      "Width": 1920,
      "Height": 1080,
      "Frequency": 15,
      "JpegQuality": 75,
      "FieldOfView": 10,
      "MinDistance": 0.1,
      "MaxDistance": 1000,
      "Topic": "/apollo/sensor/camera/front_12mm/image/compressed"
    },
    "transform": {
      "x": 0,
      "y": 1.7,
      "z": -0.2,
      "pitch": -4,
      "yaw": 0,
      "roll": 0
    }
  },

Martin20, 2020-04-17:
Hello @ntutangyun,

I think the camera is included. You can see the jetson file as below.

Moreover, I attached the cyber_monitor screenshot and the error after using the launch command.

[{"type": "CAN-Bus", "name": "CAN Bus",
            "params": {"Frequency": 10, "Topic": "/apollo/canbus/chassis"},
            "transform": {"x": 0, "y": 0, "z": 0, "pitch": 0, "yaw": 0, "roll": 0}},{"type": "GPS Device", "name": "GPS",
            "params": {"Frequency": 12.5, "Topic": "/apollo/sensor/gnss/best_pose", "Frame": "gps"},
            "transform": {"x": 0, "y": 0, "z": -1.348649, "pitch": 0, "yaw": 0, "roll": 0}},{"type": "GPS Odometry", "name": "GPS Odometry",
            "params": {"Frequency": 12.5, "Topic": "/apollo/sensor/gnss/odometry", "Frame": "gps"},
            "transform": {"x": 0, "y": 0, "z": -1.348649, "pitch": 0, "yaw": 0, "roll": 0}},{"type": "GPS-INS Status", "name": "GPS INS Status",
            "params": {"Frequency": 12.5, "Topic": "/apollo/sensor/gnss/ins_stat", "Frame": "gps"},
            "transform": {"x": 0, "y": 0, "z": -1.348649, "pitch": 0, "yaw": 0, "roll": 0}},{"type": "IMU", "name": "IMU",
            "params": {"Topic": "/apollo/sensor/gnss/imu", "Frame": "imu", "CorrectedTopic": "/apollo/sensor/gnss/corrected_imu", "CorrectedFrame": "imu"},
            "transform": {"x": 0, "y": 0, "z": -1.348649, "pitch": 0, "yaw": 0, "roll": 0}},{"type": "Radar", "name": "Radar",
               "params": {"Frequency": 13.4, "Topic": "/apollo/sensor/conti_radar"},
               "transform": {"x": 0, "y": 0.689, "z": 2.272, "pitch": 0, "yaw": 0, "roll": 0}},{"type": "Lidar", "name": "Lidar",
            "params": {"LaserCount": 32, "MinDistance": 0.5, "MaxDistance": 100, "RotationFrequency": 10, "MeasurementsPerRotation": 360, "FieldOfView": 41.33, "CenterAngle": 10, "Compensated": true, "PointColor": "#ff000000", "Topic": "/apollo/sensor/lidar128/compensator/PointCloud2", "Frame": "velodyne"},
            "transform": {"x": 0, "y": 2.312, "z": -0.3679201, "pitch": 0, "yaw": 0, "roll": 0}},{"type": "Color Camera", "name": "Main Camera",
            "params": {"Width": 1920, "Height": 1080, "Frequency": 15, "JpegQuality": 75, "FieldOfView": 50, "MinDistance": 0.1, "MaxDistance": 1000, "Topic": "/apollo/sensor/camera/front_6mm/image/compressed"},
            "transform": {"x": 0, "y": 1.7, "z": -0.2, "pitch": 0, "yaw": 0, "roll": 0}},{"type": "Color Camera", "name": "Telephoto Camera",
            "params": {"Width": 1920, "Height": 1080, "Frequency": 15, "JpegQuality": 75, "FieldOfView": 10, "MinDistance": 0.1, "MaxDistance": 1000, "Topic": "/apollo/sensor/camera/front_12mm/image/compressed"},
            "transform": {"x": 0, "y": 1.7, "z": -0.2, "pitch": -4, "yaw": 0, "roll": 0}},{"type": "Keyboard Control", "name": "Keyboard Car Control"},{"type": "Wheel Control", "name": "Wheel Car Control"},{"type": "Vehicle Control", "name": "Apollo Car Control",
            "params": {"Topic": "/apollo/control"} }]

[image]

[image]

Martin20, 2020-04-17:
Hi @EricBoiseLGSVL ,

I think the color camera is included in the configuration file. You can see below. Please let me know if it is not correct.
[{"type": "CAN-Bus", "name": "CAN Bus",
            "params": {"Frequency": 10, "Topic": "/apollo/canbus/chassis"},
            "transform": {"x": 0, "y": 0, "z": 0, "pitch": 0, "yaw": 0, "roll": 0}},{"type": "GPS Device", "name": "GPS",
            "params": {"Frequency": 12.5, "Topic": "/apollo/sensor/gnss/best_pose", "Frame": "gps"},
            "transform": {"x": 0, "y": 0, "z": -1.348649, "pitch": 0, "yaw": 0, "roll": 0}},{"type": "GPS Odometry", "name": "GPS Odometry",
            "params": {"Frequency": 12.5, "Topic": "/apollo/sensor/gnss/odometry", "Frame": "gps"},
            "transform": {"x": 0, "y": 0, "z": -1.348649, "pitch": 0, "yaw": 0, "roll": 0}},{"type": "GPS-INS Status", "name": "GPS INS Status",
            "params": {"Frequency": 12.5, "Topic": "/apollo/sensor/gnss/ins_stat", "Frame": "gps"},
            "transform": {"x": 0, "y": 0, "z": -1.348649, "pitch": 0, "yaw": 0, "roll": 0}},{"type": "IMU", "name": "IMU",
            "params": {"Topic": "/apollo/sensor/gnss/imu", "Frame": "imu", "CorrectedTopic": "/apollo/sensor/gnss/corrected_imu", "CorrectedFrame": "imu"},
            "transform": {"x": 0, "y": 0, "z": -1.348649, "pitch": 0, "yaw": 0, "roll": 0}},{"type": "Radar", "name": "Radar",
               "params": {"Frequency": 13.4, "Topic": "/apollo/sensor/conti_radar"},
               "transform": {"x": 0, "y": 0.689, "z": 2.272, "pitch": 0, "yaw": 0, "roll": 0}},{"type": "Lidar", "name": "Lidar",
            "params": {"LaserCount": 32, "MinDistance": 0.5, "MaxDistance": 100, "RotationFrequency": 10, "MeasurementsPerRotation": 360, "FieldOfView": 41.33, "CenterAngle": 10, "Compensated": true, "PointColor": "#ff000000", "Topic": "/apollo/sensor/lidar128/compensator/PointCloud2", "Frame": "velodyne"},
            "transform": {"x": 0, "y": 2.312, "z": -0.3679201, "pitch": 0, "yaw": 0, "roll": 0}},{"type": "Color Camera", "name": "Main Camera",
            "params": {"Width": 1920, "Height": 1080, "Frequency": 15, "JpegQuality": 75, "FieldOfView": 50, "MinDistance": 0.1, "MaxDistance": 1000, "Topic": "/apollo/sensor/camera/front_6mm/image/compressed"},
            "transform": {"x": 0, "y": 1.7, "z": -0.2, "pitch": 0, "yaw": 0, "roll": 0}},{"type": "Color Camera", "name": "Telephoto Camera",
            "params": {"Width": 1920, "Height": 1080, "Frequency": 15, "JpegQuality": 75, "FieldOfView": 10, "MinDistance": 0.1, "MaxDistance": 1000, "Topic": "/apollo/sensor/camera/front_12mm/image/compressed"},
            "transform": {"x": 0, "y": 1.7, "z": -0.2, "pitch": -4, "yaw": 0, "roll": 0}},{"type": "Keyboard Control", "name": "Keyboard Car Control"},{"type": "Wheel Control", "name": "Wheel Car Control"},{"type": "Vehicle Control", "name": "Apollo Car Control",
            "params": {"Topic": "/apollo/control"} }]

ntutangyun, 2020-04-18:
according to configs in JSON, both cameras are publishing at frequency 15Hz. But your `cyber_monitor` shows they are publishing at 60Hz. Something is wrong..

Martin20, 2020-04-18:
@ntutangyun Thank you,

any suggestions?

EricBoiseLGSVL, 2020-04-20:
@Martin20 Yes, you have the correct sensor.  @luqiang21 Do you know what could be the issue?

luqiang21, 2020-04-22:
@Martin20 The error on your screen seems saying traffic lights cannot be projected on the image. But I checked your configuration for cameras, looks like you are using the sample configurations. Are you using map from our simulator or your own map? Is your perception working well?

Martin20, 2020-04-22:
Actually, the Module Delay in dreamview shows less than 1 second for Traffic Light. After a while, when the camera view is changed, the delay constantly increases. I think the data from camera has a problem.

Martin20, 2020-04-22:
Hi @luqiang21,

I used your map. CubeTown. The perception works well. Just I have a problem with the traffic lights.

luqiang21, 2020-04-22:
@Martin20 That's very weird. I encountered situation that perception doesn't work but traffic light was working because GPU memory was almost exhausted. Did you try to only enable traffic light without perception?

Martin20, 2020-04-23:
@luqiang21
When I turn the perception off, the traffic light detection does not work. I think it is related to the camera view. When the vehicle encounters a new scene, the traffic light module becomes off.

Martin20, 2020-04-23:
@luqiang21 
Might it be because of the old version of the simulator? I use 2019.10 on windows. I use the old version because I have a problem with the latest version. I have an open issue for that [here](https://github.com/lgsvl/simulator/issues/726#issue-600987995).

luqiang21, 2020-04-23:
@Martin20 It is possible due to the version. But I am not clear what do you mean by "When the vehicle encounters a new scene", what is the scene here？

EricBoiseLGSVL, 2020-10-20:
We have found that sun is causing an issue with traffic light detection on Cube Town.  We are working on a fix.  In the meantime, increase the emission values for the lights or change the time of day.  We hope to have better traffic lights soon

closed 2020-10-20 by EricBoiseLGSVL

Martin20, 2020-10-21:
I think this problem is not related to a specific map. It is in the other maps as well.

EricBoiseLGSVL, 2020-10-21:
Yes, any place the sun shines on the lights.  We are working on an improved light system and should resolve this problem.  We hope to have it next release

Martin20, 2020-10-21:
so, you mean it would work if I set the simulation time to midnight?

I don't think it changes anything

EricBoiseLGSVL, 2020-10-21:
Well looking at the history, it says you are using the older version of the simulator.  Have you not been able to update to 2020.06?

kuyun-drimaes, 2020-11-10:
Traffic light module need "/apollo/sensor/camera/front_6mm/image,/apollo/sensor/camera/front_12mm/image" topics.
But LGSVL generate ""/apollo/sensor/camera/front_6mm/image/compressed,/apollo/sensor/camera/front_12mm/image/compressed" topics."
So trffic light module can't operation.

Martin20, 2020-11-10:
do you know how I can fix it?

best regards,

EricBoiseLGSVL, 2020-11-10:
We are working on new signal meshes with new logic.  This should release soon

kuyun-drimaes, 2020-11-10:
You must add "Sensor JSON" code in Vehicle setting.(Ref:
https://www.lgsvlsimulator.com/docs/modular-testing/)
And then don't start traffic light module.

I think LGSVL's main role is simulation for perception modules.
The trffaic light module is one of perception.