lgsvl/simulator #728: Static Obstacle and Traffic Light

ozguradas, 2020-04-19:
Hello LGSVL team,

I have a unity package for a map which is not created by me, so i don't have so much experinece on unity. I want to add traffic lights on that map and for sure I want to be able to control that lights. And also I need a static obstacle on the road, and I would change it's existince time to time. So let's say when i checked the traffic box on Simulator I want to see that static obstacle (it can be car) on the road. I'm using lgsvl-autoware connection btw.

So what is the easist way to do those things that I talked about ? Can you help me with that. Thank you.

EricBoiseLGSVL, 2020-04-20:
@ozguradas Thank you for using LGSVL Simulator and welcome to Unity development.  What you are asking for is an advanced topic but let's see if I can get you started in the right direction.
- First thing is to watch some tutorials on Unity especially on how to use the Editor and prefabs.
- Then look at our documentation [here](https://www.lgsvlsimulator.com/docs/). Bookmark it
- Watch [this](https://youtu.be/S0w2ahhEPbE) video.  This is a step by step guide on how to make a Unity Asset Store map into a LGSVL Simulator map.
- Then clone CubeMap repo into your project and see how the traffic light is working (you can even make a prefab of the entire system and bring into the new scene)
- To control statics and traffic lights, you will need to use the Python API, refer [here](https://www.lgsvlsimulator.com/docs/python-api/)
- For the static car, create a Controllable and for lights, we already have API control exposed.

Good luck and post any issues.

ozguradas, 2020-04-21:
Thank you Eric, i'll study on these

closed 2020-10-20 by EricBoiseLGSVL