lgsvl/simulator #729: 2019.1.10f1 load new assetbundle error

yujie181152, 2020-04-20:
hi there, i wanne know that if i can use old released simulator source to load new assetbundle which from website downloaded ,for example,i use 2019.10f1 version to load the latest assetbundle ,is it ok?

i have tried to use 2019.10f1 build assetbundle  it shows "ouf of data ",and after that i  also have tried to load latest assetbundle from website but shows "Invalid: (Line: 1, Col: 1, Idx: 0) - (Line: 1, Col: 1, Idx: 0): Exception during deserialization".  do i have to update 2019.10f1 to 2019.3.3f1?

EricBoiseLGSVL, 2020-04-20:
@yujie181152 Sorry, Unity does not support asset bundles from older versions of Unity.  You will need to update.

yujie181152, 2020-04-21:
can you provide the old assetbundle which compatiable with 2019.10f1,
or the asset file i can build myself,because old assetbundle cache was deleted by mistake,now i'm stuck with this issue,thank you very much

EricBoiseLGSVL, 2020-04-21:
@yujie181152 You can just delete data.db and backup.db from the persistent data path [here](https://docs.unity3d.com/ScriptReference/Application-persistentDataPath.html)
On windows it is C:\Users\[YOU]\AppData\LocalLow\LG Silicon Valley Lab\LGSVL Simulator\
Once you delete these files, run the 2019.10 build and all assets will download again

EricBoiseLGSVL, 2020-05-14:
@yujie181152 Did this solve your issue?

closed 2020-07-31 by lemketron