lgsvl/simulator #730: API usage problems

xiao-hua-sheng, 2020-04-20:
I want to load the map through the API, but it has been unsuccessful and has not reported an error. It's like it's stuck

EricBoiseLGSVL, 2020-04-20:
@xiao-hua-sheng What map are you trying to load?  Is there any errors in what you are using to run the Python API?  What version of simulator and OS do you have?

xiao-hua-sheng, 2020-04-21:
I ran routine 05-ego-drive-in-circle.py, and there was no error. Using Windows 10, simulator version is 2020.03

xiao-hua-sheng, 2020-04-21:
Hello, I changed the simulator version to 2019.11. The following error occurred:Exception: Environment 'Shalun' is not available

These pictures show my settings。

EricBoiseLGSVL, 2020-04-21:
@xiao-hua-sheng not seeing any pictures.  If you are using 2020.03 you will need updated asset bundles.  That is why it isn't loading with the API,  Please delete data.db and backup.db here, C:\Users\[YOU]\AppData\LocalLow\LG Silicon Valley Lab\LGSVL Simulator on windows.  Then run the binary and all correct bundles will download.  Each release has corresponding bundles for that release

Vicky-Raymond, 2020-05-01:
> I want to load the map through the API, but it has been unsuccessful and has not reported an error. It's like it's stuck
@EricBoiseLGSVL 
I had the same problem. No matter which Python API I call, the computer will crash. Now , I'm back in the old version, and everything is ok.

EricBoiseLGSVL, 2020-05-01:
@Vicky-Raymond Did you update your asset bundles?

Vicky-Raymond, 2020-05-01:
> @Vicky-Raymond Did you update your asset bundles?

I'm not sure what kind of update you mean. I deleted data.db and backup.db, and then the map and vehicle were automatically downloaded again.

EricBoiseLGSVL, 2020-05-02:
@Vicky-Raymond Yes, that is what I mean.  What hardware, OS are you using?  Is it a certain map or a custom map.  Any extra information and Player.log will help.

Vicky-Raymond, 2020-05-02:
> @Vicky-Raymond Yes, that is what I mean. What hardware, OS are you using? Is it a certain map or a custom map. Any extra information and Player.log will help.
 Just now I reload version 2020.03. When I run the Python API（26-npc-trigger-waypoints）, the computer crashed.
[Player.log](https://github.com/lgsvl/simulator/files/4567021/Player.log)
[26-npc-trigger-waypoints.txt](https://github.com/lgsvl/simulator/files/4567024/26-npc-trigger-waypoints.txt)

Vicky-Raymond, 2020-05-02:
> @Vicky-Raymond Yes, that is what I mean. What hardware, OS are you using? Is it a certain map or a custom map. Any extra information and Player.log will help.

Just now I reload version 2020.03. When I run the Python API（26-npc-trigger-waypoints）, the computer crashed.
[Player.log](https://github.com/lgsvl/simulator/files/4567026/Player.log)
[26-npc-trigger-waypoints.txt](https://github.com/lgsvl/simulator/files/4567027/26-npc-trigger-waypoints.txt)

EricBoiseLGSVL, 2020-05-08:
@JackieLee-cn The log looks like you failed in the process of downloading.  Did you wait until the WebUI is finished downloading assets before running API quick script?

JackieLee-cn, 2020-05-09:
font{
        line-height: 1.6;
    }
    ul,ol{
        padding-left: 20px;
        list-style-position: inside;
    }

    
    
        Yes,  the WebUI is finished downloading assets .Thank you very much for your patience. I'm going to postpone solving this problem. Because of the epidemic , I control the computer remotely, and the computer restart brings many troubles.

closed 2020-10-06 by EricBoiseLGSVL