lgsvl/simulator #731: Main Camera Image Crop Resize for Latencyreduction using PythonAPI and Reinforcement Learning

qsengshu, 2020-04-22:
Dear LGSVL,

Im trying to crop and resize the image i get from the MainCamera in the simulation.
I sent the Image to my reinforcement learning framework calling the method Sensor.save() (Python API File: sensor.py - command: "sensor/camera/save"). My goal is to reduce latency between the running simulation and my framework to reduce training time.

I tried following things 

**concerning cropping:**

1.(File: CameraSensorBase.cs) Cover the SensorCamera.targetTexture and DistortionTexture with a rectangle with my desired width and heigth shape. (didnt work out for me is this the right approach?)

2.(File: CameraSensorBase.cs) Changed the values of the SensorParameter (Width | Height) ( 1920 | 1080 ) to testin values (1920 | 600) and nothing happened.

3..(File: CameraSensorBase.cs) I tried too implement the cropping code with C#. i got an error 
that i cant transform unity nativearray<bytes> to byte[] c#.

4. installing opencv through a package manager. doesnt work out because its  a unity project therefore i tried to add the package as a custom dll. this also didnt workout for me.

any further ideas?

**concerning resizing:**
a) couldn't find any command. 

any further ideas?

Since so many things didnt work out for me i lost overview and clarity concerning the solution.
Which of the following points do you think are promising and I should look again up and commit to for finding a solution. Do you have further ideas for solutions to my problem?

I thank you for your help and wish you good health

Best regards,
qsengshu

EricBoiseLGSVL, 2020-04-22:
@qsengshu Method 2 should be the way to reduce camera image.  We will look into why this is not working for you in sensor parameters.  @rongguodong can you test that this is not working as intended?

martins-mozeiko, 2020-04-22:
Method 2 is the way to go, but you should be changing sensor parameters in JSON configuration instead: https://www.lgsvlsimulator.com/docs/sensor-json-options/#color-camera
Values in C# code are used only during development, when you manually instantiate sensor. They are overwritten from values in sensor configuration specified in web interface.

EricBoiseLGSVL, 2020-04-22:
@qsengshu @martins-mozeiko @rongguodong Ah, yes.  I thought you were changing the values in the sensor parameters, not the code.  I see that it was in the CameraSensorBase.  Don't do this.  The Sensor parameters allow you to change the resolution in the WebUI.  Let us know if that is working as intended.

rongguodong, 2020-04-22:
I just tried to change resolution in JSON and run the sample Python script (06-save-camera-image.py). The saved images have correct resolution as set in JSON.

EricBoiseLGSVL, 2020-04-22:
Thanks for confirming

qsengshu, 2020-04-23:
@martins-mozeiko @rongguodong @EricBoiseLGSVL 
Thank you! You have helped me a lot!

Best regards,
qsengshu

closed 2020-04-23 by qsengshu