lgsvl/simulator #733: AD Car Control can't reach the ctrl_cmd values (steering angle value)

ozguradas, 2020-04-23:
Hello,
I am using Lgsvl-Autoware connection rn. I tried both pure_pursuit and mpc_follower on it. And there was a problem with ctrl_cmd. I was monitoring steering value of ctrl_cmd and AD car control's steering value on lgsvl. I saw 0.6 steering angle on ctrl_cmd but ad car control showed me values between 0.23-0.3, and as you can understand vehicle couldn't make the turn. I know vehicle has max steering values which isn't the problem in this case. I checked the mpc_follower parameters too, I couldn't find any steering limit problem on my case. But maybe you guys can direct me to find the solution. And also I know there can be delay on steering , it wasn't the case either.

-2020.01 LGSVL
-Autoware 1.13

EricBoiseLGSVL, 2020-04-23:
@ozguradas When we implemented Autoware we had to make adjustments to make the EGO drive.  These adjustments might need changed to work with 1.13.  You should look in VehicleControlSensor.cs in Autoware reader.

[image]

@luqiang21 thoughts?

ozguradas, 2020-04-23:
Thank you for quick response @EricBoiseLGSVL , i'll play with that value which you showed me. If it is  the solution i'll let you know.

ozguradas, 2020-06-11:
Hello All, as we talked before, we couldn't reach the goal steering on lgsvl. Now we understood some of the unit/ ratio differences between lgsvl and autoware. When we control the lgsvl via autoware ai it sub. vehicle_cmd which sends angular z. When we first checked the difference between vehicle_cmd and twist_raw's angular z we saw 1/2 ratio between them. We couldn't change it via VehicleControlSensor.cs so we change the angular z which is sent to LGSVL. By meaning change, we are using 7x angular z to see proper driving on LGSVL , i don't know where is the main problem but at least i can tell there is ratio problem between the signal autoware sent and lgsvl got.

Do you guys have any solution for this ?

EricBoiseLGSVL, 2020-10-20:
@hadiTab adding you to ping @cyberphysicalrobotics

momala, 2021-03-15:
> @ozguradas When we implemented Autoware we had to make adjustments to make the EGO drive. These adjustments might need changed to work with 1.13. You should look in VehicleControlSensor.cs in Autoware reader.
> 
> [image]
> 
> @luqiang21 thoughts?

Hello all, I have also this problem (autoware 1.14) especially while we need to follow a sharp trajectory. after several try, I figured out that the real steering percentage that is published in steer_pct is half of the vehicle_cmd/twist_cmd/twist/angular/z. I don't understand whats the relation between these two values. one is angular velocity and the other is the steering percentage!
 Is there any integration to drive the steering angle or it is directly equal to half of the angular velocity?
Should I change that (0.5f) factor to achieve my desire output? I am going to compare my result with our real autonomous shuttle, therefore I need to ensure the steering command. 
I appreciate it if you can help me or introduce some doc for clarification.

EricBoiseLGSVL, 2021-03-15:
yes, you will need to change the factor if the control is not working for you.  We had to make adjustments to get autoware to drive the vehicle.

momala, 2021-03-16:
> yes, you will need to change the factor if the control is not working for you. We had to make adjustments to get autoware to drive the vehicle.

Thank you @EricBoiseLGSVL for the clarification. As I understand from other answered issues, in my current setup, although there is ctrl_cmd but lgsvl is using twist_cmd of the /vehicle_cmd topic. basically, in twist_cmd we have two values, linear velocity (X) and angular velocity (Z). I am confused here by the steering conversion. Is angular velocity converted just by a factor (i.e 0.5) to steering percentage? If it is true, in reality, where have a steering dc motor that receives velocity to handle the steering position, we will face errors. I wonder if I can optimize my pure pursuit or MPC based on simulation then implement the result on the real system. 
Is this the line that Angular velocity is converted to the steering data?

`ADSteerInput = -Mathf.Clamp(data.SteerAngularVelocity.GetValueOrDefault() * 0.5f, -1f, 1f);
`

EricBoiseLGSVL, 2021-03-17:
@momala no problem, I with I could be more help with this.  yes that is the line that it is converted. ADSteerInput is converted to work with the inputs from SimulatorControls