lgsvl/simulator #734: Autoware_Auto using ROS2 and accessing /vehicle_cmd

dr563105, 2020-04-23:
I'd like to subscribe to /vehicle_cmd using a python script node to fetch the movement controls. However, I can't seems to access RawControlCommand from autoware_auto_msgs. I'm getting "No module found" error. 

Can you please help me access this ros2 topic? I basically this to fetch throttle and brake data. 

Thanks.

EricBoiseLGSVL, 2020-04-23:
@luqiang21 Do you know what the issue is?

luqiang21, 2020-04-25:
@EricBoiseLGSVL I am not sure about this. Tae @cyberphysicalrobotics probably knows more on this.

cyberphysicalrobotics, 2020-04-25:
@dr563105 Could you see the topic name (/vehicle_control_command) on command of ros2 topic list? What's the result of ros2 topic list?

Did you use the following sensor configuration json file?
https://www.lgsvlsimulator.com/docs/autoware-auto-json-example/

autoware auto msg is defined in the following:
https://gitlab.com/autowarefoundation/autoware.auto/autoware_auto_msgs/-/tree/master/autoware_auto_msgs/msg

cyberphysicalrobotics, 2020-04-25:
I mistakenly pressed close the issue. I reopened it.

dr563105, 2020-04-25:
Thank you for your reply. 

I didn't look into the sensor config page -- https://www.lgsvlsimulator.com/docs/autoware-auto-json-example/ . I just had `/vehicle_cmd` as topic in the script. In the Simulator I saw the same as topic and type as `autoware_msgs/VehicleCmd`. So just continued with that. 

```
{
 "type": "Vehicle Control", 
 "name": "Autoware Car Control",
 "params": {"Topic": "/vehicle_control_command"} 
}
```
After adding that json into the vehicles tab in the WebUI, I see `/vehicle_control_command` with type 
`autoware_auto_msgs/RawControlCommand`

Is this correct? 

On a related note, if I custom add a sensor with a custom topic but with predefined topic type. Can I subscribe or publish that topic in the python script? 

[image]

EricBoiseLGSVL, 2020-04-28:
@cyberphysicalrobotics Thanks for helping.  @dr563105 reply is correct, right?

dr563105, 2020-04-29:
An update: able to access RawControlCommand in python.

I built everything from scratch. built and local ros sourced(~/ros2_ws/install/local_setup.bash) inside the ros2_ws. I was able to access RawControlCommand. Now in terminal1 where I run (sourced melodic) "roslaunch rosbridge_server rosbridge_websocket.launch ", terminal 2(sourced Dashing) running the node and 2020.03 Simulator running, I'm getting this error 

`2020-04-29 19:33:24+0200 [-] [ERROR] [1588181604.541905]: [Client 0] subscribe: Unable to load the manifest for package autoware_msgs. Caused by: autoware_msgs`

Have you experienced this error before? Can you please help? @EricBoiseLGSVL @cyberphysicalrobotics 

When I do 
`ros2 msg list| grep "autoware_"`
I'm getting
`autoware_auto_msgs/msg/BoundingBox
autoware_auto_msgs/msg/BoundingBoxArray
autoware_auto_msgs/msg/Complex32
autoware_auto_msgs/msg/ControlDiagnostic
autoware_auto_msgs/msg/DiagnosticHeader
autoware_auto_msgs/msg/HighLevelControlCommand
autoware_auto_msgs/msg/PointClusters
autoware_auto_msgs/msg/Quaternion32
autoware_auto_msgs/msg/RawControlCommand
autoware_auto_msgs/msg/Trajectory
autoware_auto_msgs/msg/TrajectoryPoint
autoware_auto_msgs/msg/VehicleControlCommand
autoware_auto_msgs/msg/VehicleKinematicState
autoware_auto_msgs/msg/VehicleOdometry
autoware_auto_msgs/msg/VehicleStateCommand
autoware_auto_msgs/msg/VehicleStateReport
` 
the files in my msg folder.

EricBoiseLGSVL, 2020-05-01:
@cyberphysicalrobotics any idea on this issue?

dr563105, 2020-05-01:
I saw this https://github.com/lgsvl/simulator/issues/653#issuecomment-593384836 . Since it is quite similar I went ahead changed the files they made in the page. I'm yet to test the changes. 
Only difference is I'm running the script outside of the autoware container only just import that particular message. 
Also I tried using ros2-web-bridge between the Sim and the code, there was no transfer of data. Camera sensors went on increasing their count.

Quick question - if I make changes to the .cs files, should I build them in Unity or upon clicking the simulator executable, it will automatically do run time compilation?

EricBoiseLGSVL, 2020-05-02:
@dr563105 Thanks and yes you need to build a new executable with the .cs changes.

hadiTab, 2020-05-04:
> Now in terminal1 where I run (sourced melodic) "roslaunch rosbridge_server rosbridge_websocket.launch ", terminal 2(sourced Dashing) running the node and 2020.03 Simulator running, I'm getting this [...]

@dr563105 it seems like you have written a ros2 node but are trying to send and receive data with the ros1 bridge. You need to use the ros2-web-bridge here. 

Also the autoware_auto_msgs are custom message types for autoware_auto and the bridge is unaware of their definitions by default. You need to build the autoware_auto messages and source the setup.bash file in the same terminal you run the ros2-web-bridge for the bridge to be able to understand those messages.

dr563105, 2020-05-11:
@hadiTab Thanks Hadi. You're right. I was messing around ros1 and ros2 bridges and couldn't access the msgs. Now I understand how to access it properly.

closed 2020-05-11 by dr563105