lgsvl/simulator #735: Created new camera sensors, the whole UI is overbright.

dr563105, 2020-04-23:
I created three camera sensor nodes - left, main and right adjusting the **x** variable as 0, -1 and +1 in transform sensor. In the vehicles tab I used ROS2 as bridge type and these as sensor parameters. 

`[{"type": "Color Camera", "name": "Main Camera",
            "params": {"Width": 1920, "Height": 1080, "Frequency": 15, "JpegQuality": 75, "FieldOfView": 50, "MinDistance": 0.1, "MaxDistance": 1000, "Topic": "/simulator/camera_node/main/image/compressed", "Frame": "main_camera"},
            "transform": {**"x": 0,** "y": 1.7, "z": -0.2, "pitch": 0, "yaw": 0, "roll": 0}},{"type": "Color Camera", "name": "Left Camera",
            "params": {"Width": 1920, "Height": 1080, "Frequency": 15, "JpegQuality": 75, "FieldOfView": 50, "MinDistance": 0.1, "MaxDistance": 1000, "Topic": "/simulator/camera_node/left/image/compressed", "Frame": "left_camera"},
            "transform": {**"x": -1**, "y": 1.7, "z": -0.2, "pitch": 0, "yaw": 0, "roll": 0}},{"type": "Color Camera", "name": "Right Camera",
            "params": {"Width": 1920, "Height": 1080, "Frequency": 15, "JpegQuality": 75, "FieldOfView": 50, "MinDistance": 0.1, "MaxDistance": 1000, "Topic": "/simulator/camera_node/right/image/compressed", "Frame": "right_camera"},
            "transform": {**"x": 1**, "y": 1.7, "z": -0.2, "pitch": 0, "yaw": 0, "roll": 0}},{"type": "Keyboard Control", "name": "Keyboard Car Control"},{"type": "Vehicle Control", "name": "Autoware Car Control",
            "params": {"Topic": "/vehicle_cmd"} }]`

Now upon loading the simulator in WebUI, the Simulator is overbright. 

What did do wrong?

[image]

rongguodong, 2020-04-23:
Looks like related to auto exposure. I see half of the image is dark because of shadow. So auto exposure may improve the exposure to make the whole image brighter. If you drive to somewhere with less shadow, will the whole image become less bright?

However, it should not be related to how many cameras you are using. Do you mean the image is normal if you have only one main camera, but become bright after you added left and right camera?

dr563105, 2020-04-23:
I just tried it BorregasAve with no shadow, same behaviour. Happens when there is only one sensor. It was working fine until about 5 hours ago. I even deleted the LGSVL simulator config and loaded again. Over exposed UI greets me. It's weird. Logging off and back in also doesn't help.

EricBoiseLGSVL, 2020-04-24:
@dr563105 How are you disabling shadows?  2020.03 has exposure post processing volumes added to give a more realistic exposure to direct sunlight.  This or screen space reflections are causing materials to reflect in shadow.  We will look into this more.

EricBoiseLGSVL, 2020-04-24:
@rongguodong adding you and pinging Eric to look at this issue

dr563105, 2020-04-24:
@EricBoiseLGSVL I didn't do anything apart adding those camera sensors. After raising the issue, I even reset everything and tried again. No luck. All I remember is running the sim just after adding the sensor json caused this change. 

Anyway here is new logs of 2020.03 Sim. That UI asking to choose resolution and graphic quality didn't appear despite deleting the config folder. 
[Player_2020_03.log](https://github.com/lgsvl/simulator/files/4531226/Player_2020_03.log)
 
Also attaching BorresgasAve setup screenshot. 
[image]

ericSterner, 2020-04-24:
@dr563105 We're currently investigating some issues with Unity's Physically Based Sky. It seems that the ground tint setting might have something to do with what you are seeing in the SF map.  This still needs to be verified but I'm looking into it right now.

ericSterner, 2020-04-24:
[image]
Here's a comparison for a white ground tint (which solves some other issues we were having) and a brown tint based on Unity's default settings.

rongguodong, 2020-04-24:
I tested in SF scene, and verified it is because of auto exposure, and having nothing to do with number of camera sensor. Even with the default JSON setting, I can see the overbright effect. I tried to rotate view point in "free view" mode, and see the UI become normal as long as the overall brightness of the screen is normal. See attached screenshots for caparison.
[image]
[image]
[image]
[image]

EricBoiseLGSVL, 2020-04-24:
@rongguodong @ericSterner yes, nice work on finding the issue.  If you adjust volume Physically Based Sky ground tint to black it reduces this issue.  We are looking at solutions to fix

EricBoiseLGSVL, 2020-05-08:
@dr563105 we are currently looking at a new way to adjust lighting to reduce exposure post processing.  We hope to release this change soon

dr563105, 2020-05-08:
Thanks @EricBoiseLGSVL .There is also another minute glitch(bug) I found when I add JSON sensor parameters, change the bridge type in the Vehicles(say Hyundai2018Nexo) tab of WebUI and momentarily when I click 'submit', the dialogue box flickers with title to "Configuration" and Bridge type to "no bridge". 

Screenshot of the behaviour - https://gfycat.com/wideeyedhopefulgenet

closed 2020-10-14 by EricBoiseLGSVL