lgsvl/simulator #736: How Can I run the simulator on a remote server

Martin20, 2020-04-24:
I want to run lgsvl on an ubuntu remote server. However, when I run it, it shows the number of the display is 0 (as the image below). I think this is because I want to run it remotely on ubuntu. I could run it remotely on windows 10. Any solution?

[image]

lemketron, 2020-04-26:
You could try installing the vino server with instructions here:
https://linuxconfig.org/ubuntu-remote-desktop-18-04-bionic-beaver-linux

I tried that and it worked for me with TigerVNC as a client.

Alternatively you could try xRDP with XFCE as the desktop environment (like Xubuntu, as this page indicates that xRDP has problems working with Unity and GNOME desktop environment in Ubuntu):
https://medium.com/@vivekteega/how-to-setup-an-xrdp-server-on-ubuntu-18-04-89f7e205bd4e

Martin20, 2020-04-26:
Thanks @lemketron ,

I want to connect from the windows as client. Do you have a suggestion for the client-side?

lemketron, 2020-04-26:
I would suggest investigating one of the options in this list, such as UltraVNC, TightVNC, or TigerVNC: https://www.geckoandfly.com/23203/vnc-client-viewer-windows-mac-linux/

Or, if you install XFCE and xRDP on the Ubuntu side, then you could presumably just use the Windows RDP (Remote Desktop Protocol) client (MSRDC).

fgrzeszc, 2020-04-30:
@lemketron does it mean you were able to use GPU of the remote Linux host with VNC? I never succeeded and at the end we switched to using Windows for simulator for this reason.

lemketron, 2020-04-30:
@fgrzeszc I have not tried VNC to a remote Linux host but you might find [this comment](https://github.com/lgsvl/simulator/issues/708#issuecomment-620764475) helpful:

closed 2020-07-31 by lemketron