lgsvl/simulator #737: How it is possible to spawn several vehicles in Apollo 5.0

Martin20, 2020-04-28:
Hello,

I want to have several vehicles on one simulation. I could run it with two machines. However, I do not have enough machines to run for all of my required vehicles. Is there any other way to spawn vehicles inside lgsvl and control by apollo?

EricBoiseLGSVL, 2020-04-28:
@Martin20 Yes, you can select multiple vehicles in the WebUI and configure each one in the Vehicles tab.  If you want the same vehicle type, be sure to create separate Vehicle configurations.  You can also run a distributed simulation with multiple EGO vehicles because your sensor configuration may need multiple GPUs for performance reasons.  This feature is still in development and we will be releasing more information on it soon.  We recommend you use multiple separate machines to run Apollo because of the resources needed to run both simulator and Apollo.

Martin20, 2020-04-28:
@EricBoiseLGSVL 
As I said, I do not have enough separated machines. Form your first part of explanations, I did not understand how I can have multiple vehicles on the same machine.

lemketron, 2020-04-28:
Running the full Apollo (5.0+) autonomous driving stack requires a pretty powerful machine on its own. Therefore running a single Apollo instance along with the simulator on a single machine can be challenging, but it is doable.  However, I don't believe that running multiple instances of Apollo along with the simulator on a single machine is practical.

As @EricBoiseLGSVL said, the simulator DOES support having multiple ego vehicles, as well as dozens of NPC vehicles which are "driven" by an internal NPC controller provided by the simulator. But each configured ego vehicle will typically have a set of sensors (which require simulator computation to render) and a bridge (e.g. CyberRT) with a specified IP address that can be localhost if the corresponding Apollo stack is on the same machine, or a remote machine if it's going to run on another machine.

That is, if you want two ego vehicles controlled by Apollo, you will likely need two machines configured with Apollo, and you would create two ego vehicle instances in the simulator settings and identify the IP address and port of the bridge for an Apollo instance to control each ego. While you could attempt run all of this on a single machine, if you are enabling perception and prediction as well as planning and control, the results may be disappointing. Without knowing exactly what you are trying to do, it's hard to say (without trying) whether it will work for you.

Keep in mind that in an actual vehicle, Apollo 5.0 requires either an extremely powerful computer or two separate industrial-grade and still reasonably powerful computers (Spectra, as documented in the Apollo hardware requirements).

Finally, if you're wanting to test specific cases where the second vehicle does not need to be driven by Apollo, you might consider controlling an NPC (non-player character) vehicle using our Python API. In that case you would only need one Apollo instance to control a single ego vehicle.

Martin20, 2020-05-01:
Thank you @lemketron 
Actually, I need to run multiple autonomous vehicles that have the full stack of apollo and have the communication through the V2X module. As I understand, you just suggest I have different machines for each vehicle. Right? If so, my simulation is limited to the number of machines I have.

lemketron, 2020-05-10:
Yes, as mentioned [above](https://github.com/lgsvl/simulator/issues/737#issuecomment-620854334): "If you want two ego vehicles controlled by Apollo, you will likely need two machines configured with Apollo, and you would create two ego vehicle instances in the simulator settings and identify the IP address and port of the bridge for an Apollo instance to control each ego. While you could attempt run all of this on a single machine, if you are enabling perception and prediction as well as planning and control, the results may be disappointing."

closed 2020-05-10 by lemketron