lgsvl/simulator #738: doesn't pick intensity from road marking except "BorregasAve".

msc9533, 2020-04-28:
Hi, thanks for making this awesome simulator!
I tried to record lidar data. but some problem with getting intensity.
I modified [this](https://github.com/lgsvl/PythonAPI/blob/master/quickstart/07-save-lidar-point-cloud.py) code and visualized recorded lidar data on Rviz, when using `current_scene` as `BorregasAve`  looks no problem with getting intensity.
[image]
but scene from `SanFrancisco`, has no difference with road surface and road marking surface.
I already tried update my enviroments from [this page](https://content.lgsvlsimulator.com/)..
[image]

Are there any solutions for this?

rongguodong, 2020-04-28:
Lidar intensity is set here (https://github.com/lgsvl/simulator/blob/master/Assets/Shaders/LidarPass.hlsl#L48).

For road surface, we defined "_MASKMAP", so intensity value is get from the metallic value, which is from an intensity texture. The intensity texture is created by artist matching the road surface. Currently, only Borregas map has this texture set. If you want to do this for other scenes, you have to generate such an intensity texture yourself.

rongguodong, 2020-04-28:
Just double checked SF scene, and found it does have an intensity texture. There is a bug making intensity not working for SF. We are working on it and will fix it soon.

msc9533, 2020-04-29:
Thanks for answer!

So, if I disable that `_MASKMAP`, then is it returned value from unity model's RGB or rendered RGB value include its lighting condition like camera?

rongguodong, 2020-04-29:
> Thanks for answer!
> 
> So, if I disable that `_MASKMAP`, then is it returned value from unity model's RGB or rendered RGB value include its lighting condition like camera?

It would be the average of RGB channels of albedo color, which is calculated in the [environment shadergraph](https://github.com/lgsvl/simulator/blob/master/Assets/Shaders/EnvironmentShaders/EnvironmentSimulation.shadergraph).

msc9533, 2020-04-30:
Thank you so much!

closed 2020-04-30 by msc9533