lgsvl/simulator #741: RTX 2080 Ti GPU is fine to run apollo 5.0?

Martin20, 2020-04-29:
Hello,

I am going to buy a machine with RTX 2080 Ti GPU. Since you mentioned in the link below that it is not suitable for apollo 5.0 simulation, do you have any update showing now apollo 5.0 works with that GPU?

https://github.com/lgsvl/apollo-5.0

The current version of Apollo does not support Volta and Turing architectures (this includes Titan V and RTX 2080 GPUs).

EricBoiseLGSVL, 2020-04-29:
@Martin20 Yes, Turing architecture is not supported.  We have Apollo 5.5 support on the road map which enables newer GPU architectures but that will not release for a few months.

Martin20, 2020-04-30:
Thanks @EricBoiseLGSVL ,

We are going to purchase a machine. Could you let me know what is your suggestion for the GPU part? Actually, let me know the preferable model series. We need to make sure it works with Apollo 5.0

EricBoiseLGSVL, 2020-05-01:
@Martin20 ASUS GTX1080 GPU-A8G- Gaming GPU Card from [here](https://github.com/ApolloAuto/apollo/blob/master/README.md)

Martin20, 2020-05-01:
Thanks @EricBoiseLGSVL ,

However, we do not want it is an ordinary PC. It should be a server. What is your suggestion for a more powerful GPU compatible with apollo?

Martin20, 2020-05-01:
What about NVIDIA® Quadro® RTX 5000 16.0GB GDDR6 (4xDP, 1xVirtualLink). Is this Volta and Turing? Cannot be supported by apollo-5.0?

EricBoiseLGSVL, 2020-05-02:
This card is Turing architecture.  We are currently working on supporting Apollo 5.5 that can use this GPU.

Martin20, 2020-05-02:
I think Nvidia GeForce GTX 1080 Ti is based on Pascal architecture. Right? Does it work on apollo? If so or not, do you know any better in performance? It has 11 GB of RAM. How much more is required?

EricBoiseLGSVL, 2020-05-03:
Yes, This GPU will work with Apollo 5.0.  The more ram the better.  11gb is good

Martin20, 2020-05-04:
@EricBoiseLGSVL 

Thanks. We have a K80. Does it work?

EricBoiseLGSVL, 2020-05-05:
@Martin20 We have not tested with that GPU.  Since we support integration with Apollo and not Apollo itself, I would check with Apollo/issues

closed 2020-05-08 by EricBoiseLGSVL

Martin20, 2020-06-27:
@EricBoiseLGSVL 

I found this. Does it work?

https://www.amazon.com/ASUS-ROG-STRIX-GTX1080TI-O11G-GAMING-DisplayPort-Overclocked-Graphics/dp/B06XRWYXJY/ref=pd_di_v2_sccai_6/133-3131674-6459066?_encoding=UTF8&pd_rd_i=B06XRWYXJY&pd_rd_r=3cd717aa-8df7-496b-abfa-a301f6d72efa&pd_rd_w=BE1iX&pd_rd_wg=Xfs1v&pf_rd_p=47579771-02b9-4db9-aad8-9d98476f87cf&pf_rd_r=WYXNZ71YH77T60X7G7DR&psc=1&refRID=WYXNZ71YH77T60X7G7DR

EricBoiseLGSVL, 2020-06-27:
That works and we will soon support Apollo 5.5

martins-mozeiko, 2020-06-27:
That's 1080Ti. Should work fine with 5.0 Apollo.

Martin20, 2020-06-29:
@EricBoiseLGSVL and @martins-mozeiko 

Thank you.

Martin20, 2020-08-08:
Hello @EricBoiseLGSVL,

We were not able to buy the device in the link above. We found a Turbo one as the link below. What do you think? Does it work with Apollo-5.0?

EricBoiseLGSVL, 2020-08-10:
@Martin20 Yes, Apollo 5.0 works with the Pascal architecture, which this card has.  I recommend you use newer cards and work with Apollo 5.5 but if you cannot find a newer card this will work with 5.0

Martin20, 2020-08-10:
Thank you, @EricBoiseLGSVL.

Could you suggest a newer one for 5.5?
We have two other machines that have apoll5.0. If we use 5.5 is it compatible with the ones with apollo-5.0? I want to run several vehicles that each of them is rum on a single machine.

EricBoiseLGSVL, 2020-08-10:
@Martin20 No, GPU's that are compatible with 5.5 may not be compatible with 5.0. If you want compatibility with 5.0 and 5.5 then stick with the Pascal