lgsvl/simulator #743: API usage problems

Vicky-Raymond, 2020-05-01:
Hello，LGSVL Team，
         When I ran the fifth example Python API script(05-ego-drive-in-circle.py), I only changed the vehicle name, but It was prompted with an error as shown in Figure 1 .The vehicle setting in the WebOS shows normal.
Thank you very much!

fig.1
[image]
fig.2
[image]

fig.3
[image]

dr563105, 2020-05-01:
Can you try "Jaguar2015XE" instead and see if it works?

Vicky-Raymond, 2020-05-01:
> Can you try "Jaguar2015XE" instead and see if it works?

It doesn't work.

dr563105, 2020-05-01:
Try with "Jaguar2015XE (Autoware)". It works but not with Apollo 3.0 or 5.0.

hadiTab, 2020-05-01:
@Vicky-Raymond Can you make sure you don't have a whitespace after '2015' in the name in the web UI?

Vicky-Raymond, 2020-05-01:
> @Vicky-Raymond Can you make sure you don't have a whitespace after '2015' in the name in the web UI?
I rechecked it. Both names in the webUI and in code are followed by the same whitespace . According to my understanding, as long as two names are the same(consider the whitespace ), it should be OK, but it doesn't seem to be.have you got any other suggestions?

Vicky-Raymond, 2020-05-01:
> Try with "Jaguar2015XE (Autoware)". It works also with Apollo 3.0 but not with Apollo 5.0.
The "Jaguar2015XE (Autoware)" doesn't appear in my web UI. This method doesn't work [/sad]

Vicky-Raymond, 2020-05-01:
Solution：
delete data.db and backup.db here, C:\Users[YOU]\AppData\LocalLow\LG Silicon Valley Lab\LGSVL Simulator on windows.

closed 2020-05-01 by Vicky-Raymond