lgsvl/simulator #744: Can I write a MPC control algorithm to control Egocar by the Python API?

JackieLee-cn, 2020-05-03:
[image]

JackieLee-cn, 2020-05-03:
How can I control the Egocar. Do all control operations need to be implemented by the Apollo or autoware platform?

EricBoiseLGSVL, 2020-05-06:
@JackieLee-cn Yes and no.  We have a simple tutorial for LaneFollowing but it is being updated for the new simulator currently.  See issue #698

JackieLee-cn, 2020-05-09:
font{
        line-height: 1.6;
    }
    ul,ol{
        padding-left: 20px;
        list-style-position: inside;
    }

    
    Thank you very much for your help.Looking forward to the release of the next version as soon as possible.On 5/7/2020 05:27，Eric Boise<notifications