lgsvl/simulator #747: Uneble to get can bus info in ROS 1

MaxandreOgeret, 2020-05-06:
Hello, I am trying to get the can bus info from ROS 1 using the websocket_bridge.

I added the sersors according to this piece of documentation : [can-bus](https://www.lgsvlsimulator.com/docs/sensor-json-options/#can-bus)

I started LGSVL from the docker dontainer.

And I get this error message from the ROS websocket bridge : 

```
2020-05-06 05:32:52+0300 [-] [ERROR] [1588732372.757863]: [Client 0] publish: Cannot infer topic type for topic /canbus as it is not yet advertised
```

Do you know how to solve this ? Thanks !

MaxandreOgeret, 2020-05-06:
Works when setting the topic to `/can_info`.

These sensor setting works : 

```json
{
    "type": "CAN-Bus",
    "name": "CAN Bus",
    "params": {
      "Frequency": 10,
      "Topic": "/can_info"
    },
    "transform": {
      "x": 0,
      "y": 0,
      "z": 0,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    }
}
```

Its odd though.

EricBoiseLGSVL, 2020-05-06:
@hadiTab Any idea on this issue?

hadiTab, 2020-05-06:
@MaxandreOgeret it looks like we haven't actually defined the CanBus message for ROS 1. We should have mentioned this in the documentation. As it is though, the CanBus sensor only works for ROS-Apollo and CyberRT.

MaxandreOgeret, 2020-05-07:
Ok thanks ! Any idea how to deal with this problem in an elegant way as I need the can bus data ?

tajinder96, 2020-10-05:
Any updates on this issue?

EricBoiseLGSVL, 2020-10-05:
@MaxandreOgeret You can define the CanBus in the bridge in Simulator source right @hadiTab ?

tajinder96, 2020-10-05:
@EricBoiseLGSVL I was trying this just now and I simply used the same conversion for ROS1 as defined for ROS2, and it works fine. I still need to check if any modifications to the conversion for ROS1 are required, but is there something to be particularly careful about when converting to ROS1 vs ROS2 (specifically for the can bus data message)?

EricBoiseLGSVL, 2020-10-05:
@tajinder96 Nice!  I'm not the expert on ROS, @hadiTab can let you know

hadiTab, 2020-10-05:
@tajinder96 the LGSVL message definitions are the same for ROS1 and ROS2 so message conversions will also be the same. Nothing to watch out for. Using the same conversion for ROS1 should work fine.

tajinder96, 2020-10-06:
@EricBoiseLGSVL @hadiTab Great, thanks a lot!

closed 2020-10-20 by EricBoiseLGSVL

rudolfkrecht, 2020-11-23:
Hello, could you please give more details about how it is possible to get Can Bus info with ROS 1? I was trying to modify the simulator source based on the code written for the ROS 2 bridge, without any luck. Thank you!

hadiTab, 2020-11-26:
@rudolfkrecht in `RosBridgeFactory.cs` paste the following at line 119:
```
RegPublisher<CanBusData, Lgsvl.CanBusDataRos>(plugin, Conversions.RosConvertFrom);
```

in `RosConversions.cs` add the following method to the `Conversions` class:
```
public static Lgsvl.CanBusDataRos RosConvertFrom(CanBusData data)
        {
            return new Lgsvl.CanBusDataRos()
            {
                header = new Ros.Header()
                {
                    stamp = ConvertTime(data.Time),
                    frame_id = data.Frame,
                },
                speed_mps = data.Speed,
                throttle_pct = data.Throttle,
                brake_pct = data.Braking,
                steer_pct = data.Steering,
                parking_brake_active = false,   // parking brake is not supported in Simulator side
                high_beams_active = data.HighBeamSignal,
                low_beams_active = data.LowBeamSignal,
                hazard_lights_active = data.HazardLights,
                fog_lights_active = data.FogLights,
                left_turn_signal_active = data.LeftTurnSignal,
                right_turn_signal_active = data.RightTurnSignal,
                wipers_active = data.Wipers,
                reverse_gear_active = data.InReverse,
                selected_gear = (data.InReverse ? Lgsvl.Gear.GEAR_REVERSE : Lgsvl.Gear.GEAR_DRIVE),
                engine_active = data.EngineOn,
                engine_rpm = data.EngineRPM,
                gps_latitude = data.Latitude,
                gps_longitude = data.Longitude,
                gps_altitude = data.Altitude,
                orientation = Convert(data.Orientation),
                linear_velocities = ConvertToVector(data.Velocity),
            };
        }
```

That should do it.

rudolfkrecht, 2020-11-26:
@hadiTab Thank you a lot for the detailed answer, now it works perfectly.