lgsvl/simulator #748: Starting ROS2 Node in LGSVL Python API

qsengshu, 2020-05-06:
Dear LGSVL,

Im running the simulation through python api. starting the simulation with sanfrancisco scence, adding the agent, setting the input action and running the simulation for 0.25s. Then the simulation gets stopped and i desire to read out the main camera image. After that i repeat those steps.

When the simulation is stopped I do not want to use the existing main camera image save command in the python api. Instead I want to read the ROS2 node via the rosbridge.

For this purpose, ros2 dashing was installed, the rosbridge was activated and a custom ros package was set up. The main camera image node is configured in the web ui. This allows to read the main camera node in the shell. 

My Question:
1. How do I get the camera image which is displayed in the shell into my python api code? so that I can edit them afterwards with an image processing library.

2. The nodes publish the images with a certain frequency. How can I only read single image frame after the simulation was stopped? Do I have to work with ros node spin_once for this?

3 Are the ego vehicle callbacks intended for my problem ? 

quote from the link (https://www.lgsvlsimulator.com/docs/python-api/) :
"""
In addition to Agent callbacks, EgoVehicle has one extra callback.
on_custom - called when a Sensor Plugin sends a callback; accepts three arguments: (agent, kind, context) - agent instance, kind of sensor plugin as string, JSON context
"""
If so, I cannot find any information about the context parameter.

Best regards,
qsengshu

EricBoiseLGSVL, 2020-05-06:
@qsengshu 1) I'm not quite sure what you are trying to do here.  The Camera sensor is saving the file to disk.  Then you can create python code to process the image(s).
2) I would create a new CameraSensor plugin that can have any custom logic you will need.
3) You could add this to the APIManager.cs AddCustom Event and edit PythonAPI but as is, this is not supported.  From my response to 1) why does the API need the image after it is saved.

qsengshu, 2020-05-06:
My problem with saving and reading the image from the hard disk is that it is a slow operation. In reinforcement learning this leads to an increase in training time. That is why I want to get around this.

Thanks for the help! I'm gonna try this out.

Best regards,
qsengshu

martins-mozeiko, 2020-05-06:
You can instruct Python API to save image to memory disk. On Linux anything you save to `/dev/shm` folder, for example, `/dev/shm/image.jpg`, is not actually written to disk. It stays in memory. And saving/loading files from memory disk is very fast. If you don't want to write to /dev/shm, you can always create new mountpoint with tmpfs filesystem.

qsengshu, 2020-05-07:
That sounds like a very good solution for me in this case. Thanks a lot.

Best regards,
qsengshu

closed 2020-05-07 by qsengshu