lgsvl/simulator #753: Simulator not working

ctgloglu, 2020-05-09:
I tried to run it by following the steps [here](https://www.lgsvlsimulator.com/docs/getting-started/#downloading-and-starting-simulator) but it didn't work.I chose the nvidia driver.I install `sudo apt install libvulkan1`.Then I tried to run the simulator but the black screen turns on and off.Please help me.

lemketron, 2020-05-10:
Please check the following:
1. Review the Player.log file (~/.config/unity3d/LG Silicon Valley Lab/LGSVL Simulator/Player.log) to look for any helpful messages or errors.
2. Check the output from this: `grep ^Vulkan ~/.config/unity3d/LG\ Silicon\ Valley\ Lab/LGSVL\ Simulator/Player.log`
3. In particular, if you see "mesa" or "Intel" mentioned in Vulkan vendor or renderer lines rather than Nvidia, then you might have "mesa-vulkan-drivers" installed. If so, then you can remove these with `sudo apt remove mesa-vulkan-drivers`.
4. Try installing vulkan-tools (`sudo apt install vulkan-tools`). Type `vulkaninfo` to verify that libvulkan is working, and then see if `vkcube` displays a window with a spinning 3D cube.

ctgloglu, 2020-05-10:
~/.config/unity3d/LG Silicon Valley Lab/LGSVL Simulator/Player.log
```
Mono path[0] = '/home/can/Desktop/lgsvlsimulator-linux64-2020 (1).03/simulator_Data/Managed'
Mono config path = '/home/can/Desktop/lgsvlsimulator-linux64-2020 (1).03/simulator_Data/MonoBleedingEdge/etc'
Preloaded 'liblaszip.so'
Preloaded 'libpng.so'
Preloaded 'libsqlite3.so'
Preloaded 'libturbojpeg.so'
Display 0 '0': 1920x1080 (primary device).
Loading player data from /home/can/Desktop/lgsvlsimulator-linux64-2020 (1).03/simulator_Data/data.unity3d
Unable to load player prefs
Desktop is 1920 x 1080 @ 60 Hz
[Vulkan init] extensions: count=15
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
Vulkan detection: 0
No supported renderers found, exiting 
(Filename:  Line: 618)

```
grep ^Vulkan ~/.config/unity3d/LG\ Silicon\ Valley\ Lab/LGSVL\ Simulator/Player.log
`Vulkan detection: 0`

lemketron, 2020-05-10:
This is a problem: `No supported renderers found, exiting`

Do you have a GPU with latest NVIDIA drivers installed as described in the [getting started docs](https://www.lgsvlsimulator.com/docs/getting-started/#downloading-and-starting-simulator)?

ctgloglu, 2020-05-10:
I solved the problem by installing the PPA system. Thanks for your help.

closed 2020-05-10 by ctgloglu

komizomakoto, 2021-05-13:
I have same error with this issue.
Please help me anyone.

My environment is as follows:
OS: Ubuntu 18.04
GTX: GeForce RTX 3090
NVIDIA-driver: 460.73.01

I want to use this simulator with AutowareAuto so that I do command ```/opt/lgsvl/simulator``` in
ADE container.
But LGSVL quits immediately after starting.
I have already done command  ```sudo apt remove mesa-vulkan-drivers```.

Player.log is like below.
~/.config/unity3d/LG Silicon Valley Lab/LGSVL Simulator/Player.log
output:
```
....
Preloaded 'libvisualization_msgs_interactive_marker_control__rosidl_typesupport_introspection_c_native.so'
Preloaded 'libvisualization_msgs_interactive_marker_feedback__rosidl_typesupport_c_native.so'
Preloaded 'libvisualization_msgs_interactive_marker_feedback__rosidl_typesupport_fastrtps_c_native.so'
Preloaded 'libvisualization_msgs_interactive_marker_feedback__rosidl_typesupport_introspection_c_native.so'
Preloaded 'libvisualization_msgs_interactive_marker_init__rosidl_typesupport_c_native.so'
Preloaded 'libvisualization_msgs_interactive_marker_init__rosidl_typesupport_fastrtps_c_native.so'
Preloaded 'libvisualization_msgs_interactive_marker_init__rosidl_typesupport_introspection_c_native.so'
Preloaded 'libvisualization_msgs_interactive_marker_pose__rosidl_typesupport_c_native.so'
Preloaded 'libvisualization_msgs_interactive_marker_pose__rosidl_typesupport_fastrtps_c_native.so'
Preloaded 'libvisualization_msgs_interactive_marker_pose__rosidl_typesupport_introspection_c_native.so'
Preloaded 'libvisualization_msgs_interactive_marker_update__rosidl_typesupport_c_native.so'
Preloaded 'libvisualization_msgs_interactive_marker_update__rosidl_typesupport_fastrtps_c_native.so'
Preloaded 'libvisualization_msgs_interactive_marker_update__rosidl_typesupport_introspection_c_native.so'
Preloaded 'libvisualization_msgs_marker__rosidl_typesupport_c_native.so'
Preloaded 'libvisualization_msgs_marker__rosidl_typesupport_fastrtps_c_native.so'
Preloaded 'libvisualization_msgs_marker__rosidl_typesupport_introspection_c_native.so'
Preloaded 'libvisualization_msgs_marker_array__rosidl_typesupport_c_native.so'
Preloaded 'libvisualization_msgs_marker_array__rosidl_typesupport_fastrtps_c_native.so'
Preloaded 'libvisualization_msgs_marker_array__rosidl_typesupport_introspection_c_native.so'
Preloaded 'libvisualization_msgs_menu_entry__rosidl_typesupport_c_native.so'
Preloaded 'libvisualization_msgs_menu_entry__rosidl_typesupport_fastrtps_c_native.so'
Preloaded 'libvisualization_msgs_menu_entry__rosidl_typesupport_introspection_c_native.so'
Preloaded 'libtest_msgs__python.so'
Display 0 'rdp0': 1920x1080 (primary device).
Loading player data from /opt/lgsvl/simulator_Data/data.unity3d
Unable to load player prefs
Desktop is 1920 x 1080 @ 50 Hz
[Vulkan init] extensions: count=2
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
Vulkan detection: 0
No supported renderers found, exiting 
(Filename:  Line: 618)
```
And I also show an output of command ```vulkaninfo```. 
```
ERROR: [Loader Message] Code 0 : loader_scanned_icd_add: Could not get 'vkCreateInstance' via 'vk_icdGetInstanceProcAddr' for ICD libGLX_nvidia.so.0
Cannot create Vulkan instance.
This problem is often caused by a faulty installation of the Vulkan driver or attempting to use a GPU that does not support Vulkan.
/build/vulkan-tools-KEbD_A/vulkan-tools-1.2.131.1+dfsg1/vulkaninfo/vulkaninfo.h:371: failed with ERROR_INCOMPATIBLE_DRIVER
```
Thank you for your cooperation.

marcusvinicius178, 2021-07-30:
Hi @stiglioglu which PPA did you enabled? I believe am facing the same issue here: https://github.com/lgsvl/simulator/issues/1646

I have disabled some PPA repo when fixing my ubuntu.....now I cannot dowload any asset from SVL. Could you kindly provide the command line you have runned?

RedBlack888, 2021-08-18:
The same issue,  `sudo apt remove mesa-vulkan-drivers` has done, and follow step by step as the official website https://www.svlsimulator.com/docs/installation-guide/installing-simulator/ . 

`nvidia-smi`  as shown
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 460.91.03    Driver Version: 460.91.03    CUDA Version: 11.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  GeForce RTX 208...  Off  | 00000000:01:00.0 Off |                  N/A |
| 26%   33C    P8     2W / 260W |    171MiB / 11016MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+

+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      1640      G   /usr/lib/xorg/Xorg                 45MiB |
|    0   N/A  N/A    901777      G   /usr/lib/xorg/Xorg                 47MiB |
|    0   N/A  N/A    901907      G   /usr/bin/gnome-shell               65MiB |
+-----------------------------------------------------------------------------+

` ~/.config/unity3d/LG\ Silicon\ Valley\ Lab/LGSVL\ Simulator/Player.log` 

Vulkan driverversion=[0x7316c0c0] uint=[1930870976]
Vulkan PSO: cache data not found [path=/tmp/LGElectronics/SVLSimulator/vulkan_pso_cache.bin]
ALSA lib confmisc.c:767:(parse_card) cannot find card '0'
ALSA lib conf.c:4732:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4732:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1246:(snd_func_refer) error evaluating name
ALSA lib conf.c:4732:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5220:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM default
FMOD failed to initialize the output device.: "Error initializing output device. " (60)

Forced to initialize FMOD to to the device driver's system output rate 48000, this may impact performance and/or give inconsistent experiences compared to selected sample rate 48000

ALSA lib confmisc.c:767:(parse_card) cannot find card '0'
ALSA lib conf.c:4732:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4732:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1246:(snd_func_refer) error evaluating name
ALSA lib conf.c:4732:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5220:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM default
FMOD failed to initialize the output device.: "Error initializing output device. " (60)

FMOD failed to initialize any audio devices, running on emulated software output with no sound. Please check your audio drivers and/or hardware for malfunction.

COULD YOU TELL ME HOW TO SOLVE THIS ISSUE? 

BEST