lgsvl/simulator #754: How can I connect the ros in windows

ctgloglu, 2020-05-10:
How can I connect ros bridge to my simulation.

EricBoiseLGSVL, 2020-05-14:
@ctgloglu this is our instructions we use internally

# Installing ROS & rosbridge on Windows:
1. Install window-linux subsystem: https://docs.microsoft.com/en-us/windows/wsl/install-win10

2. Open Microsoft Store from start menu and search for ubuntu. Select Ubuntu 18.04 and install it.
   - You will be asked to setup root username and password on first startup (don't forget these!)
   - Note that installation from Microsoft Store is blocked by IT at the office.

3. Open ubuntu terminal (it should be in start menu) and install ros:

   ```
   sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'
   
   sudo apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654
   
   sudo apt update
   
   sudo apt install ros-melodic-ros-base
   
   sudo rosdep init
   
   rosdep update
   
   echo "source /opt/ros/melodic/setup.bash" >> ~/.bashrc
   
   source ~/.bashrc
   ```

4. Install rosbridge:

   ```
   sudo apt install ros-melodic-rosbridge-suite
   ```

5. To run rosbridge:

   ```
   roslaunch rosbridge_server rosbridge_websocket.launch
   ```

6. When entering rosbirdge address in the simulator use: `127.0.0.1:9090`

# Useful ROS commands:

After running the bridge open another ubuntu terminal to run ros commands in.

- To list all published topics:

  ```
  rostopic list
  ```

- To listen to a rostopic to see the data being published:

  ```
  rostopic echo /topic/name
  ```

  - replace /topic/name with the actual name of the topic
  - press Ctrl+C to stop
  - Add `-n 1` after `echo` if you only want to see one message

- You will not be able to echo messages that have custom message types. Pointcloud and camera are ros message types.

- Autoware.ai is the only autonomous stack that runs base ros 1 and would work with this bridge, when setting up the vehicle config json you can use the example for Autoware.Ai on the docs website.

Thanks @hadiTab

ctgloglu, 2020-05-15:
I ran ros bridge without any problem.Thank you for your answer.

closed 2020-05-15 by ctgloglu