lgsvl/simulator #755: How to add a new NPC to the simulator

gyp199009, 2020-05-11:
How to add a new NPC to the simulator since there are only six kinds NPCs (Sedan, BoxTruck, Jeep, SchoolBus, SUV, Hatchback) originally.

EricBoiseLGSVL, 2020-05-12:
@gyp199009 We are currently adding a feature to build external npc's and it should release soon.  If you want to add now, you can look at existing NPC prefabs as an example.

1) Edit/Create NPC model and import
Be sure to have child mesh node called Body that contains materials named:
- LightHead, LightBrake, IndicatorLeft, IndicatorRight, IndicatorReverse
Also, you need child meshes called:
- WheelLeftFront, WheelRightFront, WheelRearLeft, WheelRearRight.
Lastly you need the light holder all NPC's have to be able to toggle lights
- Duplicate the light holder object in another npc prefab and copy to your new NPC.  Move the lights to fit the new model.

2) Create empty object and name it the NPC name.

3) Place model as a child of this new object.

4) Drag to Assets/Prefabs/NPC to create a new NPC prefab

5) Select NPCManager prefab in Assets/Prefabs/Managers

6) Add a new array element for your new NPC in NPCVehicles array

7) Add the new NPC prefab to the Prefab reference and set it's type

8) Save changes to the NPCManager prefab

9) Build Simulator binary

Simulator will now load your new NPC at runtime
Note: when we release the external NPC feature, these steps will change

gyp199009, 2020-05-12:
Thanks a lot!
By the way, is it almost the same way to add a new pedestrian to the simulator by taking the existing pedstrian prefabs as examples?

EricBoiseLGSVL, 2020-05-12:
@gyp199009 exactly

closed 2020-06-18 by EricBoiseLGSVL