lgsvl/simulator #757: How can I can get the camera internal parameters like focus?

chen1qy, 2020-05-13:
I want to get the camera internal parameters of Color Camera, but I can only find the distortion parameter in JSON. Are there any default internal parameters？

rongguodong, 2020-05-13:
The distortion parameters come from calibration of real camera. We do not provide any `default` parameters.

chen1qy, 2020-05-14:
What about the internal parameters of real camera?

rongguodong, 2020-05-14:
We just provide reference implementation of camera (as well as other sensors). Users can implement their own sensor plugin matching specific real sensors they are using. We implemented a sample sensor plugin for Velodyne Lidar. More details of sensor plugin can be found at: https://www.lgsvlsimulator.com/docs/sensor-plugins/.

EricBoiseLGSVL, 2020-05-14:
@chen1qy Yes, @rongguodong is right.  If you want different camera settings, you will need to create your own camera sensor.  You can create a plugin that can be loaded without changing simulator source, or you can add to the simulator source code/sensor list and re-compile simulator.

closed 2020-06-12 by chen1qy