lgsvl/simulator #758: QUESTION on LGSVL documentation with Autoware.AI

SenRamakri, 2020-05-14:
Hello,

Im trying to get Autoware(version 1.13.0) running with LGSVL. Initially I was trying to follow the documentation here - https://www.lgsvlsimulator.com/docs/autoware-instructions/
And part of the documentation provides a sample json file at - https://www.lgsvlsimulator.com/docs/autoware-json-example/

But when I look at autoware.ai repo it has sample.json for "LGSVL Simulator" at - https://gitlab.com/autowarefoundation/autoware.ai/simulation/-/blob/master/lgsvl_simulator_bridge/config/sample.json

Why is the sample.json in Autoware.AI repo looks so different from what is mentioned in - https://www.lgsvlsimulator.com/docs/autoware-json-example/

Are these JSON files used for different purposes? How and Where are they used? Please share if you have more info on these.

Thanks in advance.

EricBoiseLGSVL, 2020-05-14:
@SenRamakri The Autoware.AI configuration looks outdated.  @hadiTab ?

SenRamakri, 2020-05-14:
> @SenRamakri The Autoware.AI configuration looks outdated. @hadiTab ?

Thanks for your quick reply @EricBoiseLGSVL. I will wait for @hadiTab to respond as well.

dr563105, 2020-05-14:
The example in autoware.ai is outdated and for the older Simulator version. 

I have tried using the ones in the lgsvl documentation site, they work(don’t know if they need to be updated). You can compare them with the json already included in the vehicles tabs with autoware in web ui.

hadiTab, 2020-05-14:
Yup, the autoware.ai JSON is outdated. Please use the one in our documentation.

SenRamakri, 2020-05-14:
Thanks for the answers everyone, I'll use the one from LGSVL documentation.

closed 2020-05-14 by EricBoiseLGSVL