lgsvl/simulator #759: Failed to build WebUI

cjonesy20, 2020-05-14:
Following the build instructions here on Windows. Unity 2019.3.3f1. 
https://www.lgsvlsimulator.com/docs/build-instructions/

Win32Exception: The system cannot find the file specified.

System.Diagnostics.Process.StartWithShellExecuteEx (System.Diagnostics.ProcessStartInfo startInfo) (at <ae22a4e8f83c41d69684ae7f557133d9>:0)
System.Diagnostics.Process.Start () (at <ae22a4e8f83c41d69684ae7f557133d9>:0)
(wrapper remoting-invoke-with-check) System.Diagnostics.Process.Start()
Simulator.Editor.BuildWebUI.Build () (at Assets/Scripts/Editor/BuildWebUI.cs:32)
UnityEngine.Debug:LogException(Exception)
Simulator.Editor.BuildWebUI:Build() (at Assets/Scripts/Editor/BuildWebUI.cs:39)

When I run npm run pack I get the following:

C:\Users\cjone31\Documents\School\Masters Computer Science\IndStudy\simulator\WebUI>npm run pack

> simulator-web-app@1.0.0 pack C:\Users\cjone31\Documents\School\Masters Computer Science\IndStudy\simulator\WebUI
> enact pack

Creating a development build...
        5.27 MB         dist\main.js
        36.17 KB        dist\main.css

Compiled with warnings:

./src/components/SimulationManager/FormTraffic.js
  Line 18:  React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?  react-hooks/exhaustive-deps
  Line 27:  React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?  react-hooks/exhaustive-deps
  Line 28:  React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?  react-hooks/exhaustive-deps
  Line 29:  React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?  react-hooks/exhaustive-deps

./src/components/SimulationManager/FormGeneral.js
  Line 23:  React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?                                                                                                                                                                    react-hooks/exhaustive-deps
  Line 24:  React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?                                                                                                                                                                    react-hooks/exhaustive-deps
  Line 25:  React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?                                                                                                                                                                    react-hooks/exhaustive-deps
  Line 26:  React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?                                                                                                                                                                    react-hooks/exhaustive-deps
  Line 30:  Assignments to the 'unmounted' variable from inside React Hook useEffect will be lost after each render. To preserve the value over time, store it in a useRef Hook and keep the mutable value in the '.current' property. Otherwise, you can move this variable directly inside useEffect  react-hooks/exhaustive-deps

./src/components/SimulationManager/FormMapVehicles.js
  Line 26:   React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?  react-hooks/exhaustive-deps
  Line 27:   React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?  react-hooks/exhaustive-deps
  Line 28:   React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?  react-hooks/exhaustive-deps
  Line 46:   React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?  react-hooks/exhaustive-deps
  Line 126:  'vehicleList' is already declared in the upper scope                                                                      no-shadow

./src/components/SimulationManager/FormWeather.js
  Line 27:  React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?  react-hooks/exhaustive-deps
  Line 32:  React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?  react-hooks/exhaustive-deps
  Line 41:  React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?  react-hooks/exhaustive-deps
  Line 50:  React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?  react-hooks/exhaustive-deps
  Line 59:  React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?  react-hooks/exhaustive-deps

./src/components/MapManager/MapManager.js
  Line 33:   React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?                                                                                                                                                                    react-hooks/exhaustive-deps
  Line 34:   React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?                                                                                                                                                                    react-hooks/exhaustive-deps
  Line 61:   Assignments to the 'unmounted' variable from inside React Hook useEffect will be lost after each render. To preserve the value over time, store it in a useRef Hook and keep the mutable value in the '.current' property. Otherwise, you can move this variable directly inside useEffect  react-hooks/exhaustive-deps
  Line 81:    React Hook useEffect has a missing dependency: 'fetchData'. Either include it or remove the dependency array                                                                                                                                                                                react-hooks/exhaustive-deps
  Line 222:  Expected '===' and instead saw '=='                                                                                                                                                                                                                                                         eqeqeq

./src/components/VehicleManager/VehicleManager.js
  Line 47:   React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?                                                                                                                                                                    react-hooks/exhaustive-deps
  Line 48:   React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?                                                                                                                                                                    react-hooks/exhaustive-deps
  Line 49:   React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?                                                                                                                                                                    react-hooks/exhaustive-deps
  Line 50:   React Hook useCallback does nothing when called with only one argument. Did you forget to pass an array of dependencies?                                                                                                                                                                    react-hooks/exhaustive-deps
  Line 75:   Assignments to the 'unmounted' variable from inside React Hook useEffect will be lost after each render. To preserve the value over time, store it in a useRef Hook and keep the mutable value in the '.current' property. Otherwise, you can move this variable directly inside useEffect  react-hooks/exhaustive-deps
[... cut ...]
  Line 62:  Expected '===' and instead saw '=='  eqeqeq

./src/components/SimulationManager/SimulationManager.js
  Line 63:  Assignments to the 'unmounted' variable from inside React Hook useEffect will be lost after each render. To preserve the value over time, store it in a useRef Hook and keep the mutable value in the '.current' property. Otherwise, you can move this variable directly inside useEffect  react-hooks/exhaustive-deps

NOTICE: This build contains debugging functionality and may run slower than in production mode.

cjonesy20, 2020-05-14:
When I click Simulator then Check in Unity I get this the following, not sure if it's relevant. 

Checking...
ERROR: Folder 'HDRPDefaultResources' should not be inside of '/Assets'
ERROR: File 'ExampleFMU.fmu' with '.fmu' extension is not allowed inside '/Assets/Resources'
ERROR: File 'SimulatorShaderVariants.shadervariants' with '.shadervariants' extension is not allowed inside '/Assets/Shaders'
ERROR: File '/Assets/Scripts/Dynamics/FMU/FMU.cs' does not have correct copyright header
ERROR: File '/Assets/Scripts/Editor/FMUEditor/FMUImporter.cs' does not have correct copyright header
ERROR: File '/Assets/Scripts/Editor/PointCloud/PointCloudImportAxes.cs' does not have correct copyright header
WARNING: File '/Assets/Scripts/PointCloud/PointCloudManager.cs' starts with non-ASCII characters, check if you need to remove UTF-8 BOM
ERROR: File '/Assets/Scripts/Utilities/Manifest.cs' does not have correct copyright header
ERROR: File name 'Foliage Diffusion Profile.asset' contains spaces in '/Assets/HDRPDefaultResources'
ERROR: File name 'Foliage Diffusion Profile.asset.meta' contains spaces in '/Assets/HDRPDefaultResources'
ERROR: File name 'Scene PostProcess Profile.asset' contains spaces in '/Assets/HDRPDefaultResources'
ERROR: File name 'Scene PostProcess Profile.asset.meta' contains spaces in '/Assets/HDRPDefaultResources'
ERROR: File name 'Skin Diffusion Profile.asset' contains spaces in '/Assets/HDRPDefaultResources'
ERROR: File name 'Skin Diffusion Profile.asset.meta' contains spaces in '/Assets/HDRPDefaultResources'
ERROR: File name 'Sky and Fog Settings Profile.asset' contains spaces in '/Assets/HDRPDefaultResources'
ERROR: File name 'Sky and Fog Settings Profile.asset.meta' contains spaces in '/Assets/HDRPDefaultResources'
Done!

eniseiskaya, 2020-05-14:
I have the same problem. I try `npm start`, `npm update`, `npm run pack`, `npm run pack -p`. So it still doesn't work.

C:\projects\simulator\WebUI>npm start
npm ERR! missing script: start

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\enise\AppData\Roaming\npm-cache\_logs\2020-05-14T10_37_35_555Z-debug.log

EricBoiseLGSVL, 2020-05-14:
@cjonesy20 @eniseiskaya The first post says "Compiled with warnings:" Which means it compiled but with warnings, which is expected.
The second is a check for when you are creating user assets to see if you are naming/placing/including and is mostly for internal development.

Steps to try for Windows
1 - Navigate to C:\Users\You\Documents\Repos\Simulator\WebUI folder
2 - Hold Shift + Right click in the folder and choose Open PowerShell window here
3 - Type npm install
4 - Type npm pack
5 - If you can't run any npm command you need Node.js [here](https://nodejs.org/en/) and make sure you have git lfs installed before cloning simulator source

eniseiskaya, 2020-05-14:
@EricBoiseLGSVL i did it but it doesn't work. Node.js. git lfs installed.

EricBoiseLGSVL, 2020-05-14:
@eniseiskaya Again Compiled with warnings is fine.  Why are you running npm start? It should be
3 - Type npm install
4 - Type npm pack
And then you are finished.  You can then run webui in editor from loader scene.

eniseiskaya, 2020-05-14:
@EricBoiseLGSVL I did as you wrote but there was still a written error

martins-mozeiko, 2020-05-14:
You do not need to run "npm start". Not sure where you read it to run "start", but it not needed. This is not node.js server or application. You need only to "pack" files so they are usable by web page later.

cjonesy20, 2020-05-14:
npm works in cmd.exe, but not in Powershell. I ran repair on NodeJS. 

I went into the registry editor and changed the path to include C:\Program Files\nodejs and then confirmed the path changed in Powershell. npm still won't run unless I go to the nodejs folder. 

`PS C:\Users\cjone31\Documents\School\Masters Computer Science\IndStudy\simulator> (Get-ItemProperty -Path 'Registry::HKEY_LOCAL_MACHINE\System\CurrentControlSet\Control\Session Manager\Environment' -Name PATH).path
C:\Python38\Scripts\;C:\Python38\;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.0\bin;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.0\libnvvp;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files\Amazon\AWSCLI\bin\;C:\Program Files\Git\cmd;C:\Program Files\nodejs\;C:\ProgramData\chocolatey\bin;;C:\Program Files\Git LFS;C:\Program Files\nodejs
PS C:\Users\cjone31\Documents\School\Masters Computer Science\IndStudy\simulator> npm
npm : The term 'npm' is not recognized as the name of a cmdlet, function, script file, or operable program. Check the
spelling of the name, or if a path was included, verify that the path is correct and try again.
At line:1 char:1
+ npm
+ ~~~
    + CategoryInfo          : ObjectNotFound: (npm:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException`

I then ran 'C:\Program Files\nodejs\npm' install and pack and it ran. Tried to Build WebUI again, same message.

eniseiskaya, 2020-05-14:
@martins-mozeiko I understood about `npm start` but first I tried to do all the instructions in order as @EricBoiseLGSVL  wrote and I checked all the components from
> 5 - If you can't run any npm command you need Node.js here and make sure you have git lfs installed before cloning simulator source

martins-mozeiko, 2020-05-14:
What error exactly are you getting now?

eniseiskaya, 2020-05-14:
@martins-mozeiko 

> Failed to build WebUI. Please make sure npm is installed. Check README.md file
UnityEngine.Debug:LogError(Object)

> Simulator.Editor.BuildWebUI:Build() (at Assets/Scripts/Editor/BuildWebUI.cs:40)

martins-mozeiko, 2020-05-14:
That means either you have installed npm without adding it to PATH. And Unity cannot find where is npm executable.
Or it could mean that you have not restarted Unity + Unity Hub after installing npm. Process picks up new PATH only at startup, not afterwards.

But - if you have run npm run pack in command-line, there is not need to run Simulator -> Build Web ui menu item. It does exactly same thing. You can skip doing that if build in command-line already succeeded.

cjonesy20, 2020-05-21:
I created a new user in Windows for an unrelated issue and when I reran npm install and pack in Powershell it installed. Then I went back to Unity and ran Build WebUI and it worked.

closed 2020-05-28 by EricBoiseLGSVL