lgsvl/simulator #760: Unable to start binary executable on linux...

jonndoe, 2020-05-14:
(base) latin@latin:~/PycharmProjects/lgsvlsimulator-linux64-2020.03$ ./simulator
Set current directory to /home/latin/PycharmProjects/lgsvlsimulator-linux64-2020.03
Found path: /home/latin/PycharmProjects/lgsvlsimulator-linux64-2020.03/simulator
(base) latin@latin:~/PycharmProjects/lgsvlsimulator-linux64-2020.03$ 

So current directory is correct as I see...Could you advice??

jonndoe, 2020-05-14:
`(base) latin@latin:~/PycharmProjects/lgsvlsimulator-linux64-2020.03$ ./simulatorSet current directory to /home/latin/PycharmProjects/lgsvlsimulator-linux64-2020.03
Found path: /home/latin/PycharmProjects/lgsvlsimulator-linux64-2020.03/simulator
(base) latin@latin:~/PycharmProjects/lgsvlsimulator-linux64-2020.03$ 
`

jonndoe, 2020-05-14:
`(base) latin@latin:~/PycharmProjects/lgsvlsimulator-linux64-2020.03$ ./simulator
Set current directory to /home/latin/PycharmProjects/lgsvlsimulator-linux64-2020.03
Found path: /home/latin/PycharmProjects/lgsvlsimulator-linux64-2020.03/simulator
(base) latin@latin:~/PycharmProjects/lgsvlsimulator-linux64-2020.03$ 
`

EricBoiseLGSVL, 2020-05-14:
@jonndoe Please post player.log / system specs and any other info you can.  This is very difficult to understand what you are asking.  Can you not find the installed executable? Does it crash on start?  More information will help us see what the issue is and will help others that might have the same problem.

jonndoe, 2020-05-14:
Thanks for your reply, after Ive downloaded the binary for linux [https://github.com/lgsvl/simulator/releases/tag/2020.03](url), I unziped it to the download folder and cd to it and run ./simulator . After this I have issue in my shell: Set current directory to /home/latin/downloads/lgsvlsimulator-linux64-2020.03 Found path: /home/latin/downloads/lgsvlsimulator-linux64-2020.03/simulator

jonndoe, 2020-05-14:
Should I make any extra commands to run executable?

EricBoiseLGSVL, 2020-05-14:
@jonndoe Thanks for clarifying.  Have you looked through the documentation on getting started yet?  You can find it [here](https://www.lgsvlsimulator.com/docs/getting-started/).  @luqiang21 is jon missing any steps for running the executable?

jonndoe, 2020-05-14:
Thank you! Works now!

closed 2020-05-14 by jonndoe